DHDAXCW/NanoPi-R4S-R4SE #2: 温度没显示

ljtao020, 2021-02-07:
你好，刷大佬2月6号的固件，在状态栏没显示温度，有什么方法可以解决吗

DHDAXCW, 2021-02-07:
当时我考虑动态超频，怕温度会影响，就把它给删了。现在目前来说，动态超频改了很多，不受影响，那就加上去吧。大概下午就可以用了😁

ljtao020, 2021-02-08:
大佬，还没重新编译好吗？我刷了你自动编译那些也还没有显示呢

发自我的iPhone

DHDAXCW, 2021-02-08:
还在努力，今天应该用上吧

DHDAXCW, 2021-02-08:
今天编译很顺利，目录已找到了😁

DHDAXCW, 2021-02-08:
温度已加上了，祝好运！😀

ljtao020, 2021-02-09:
大佬，新版已经用上，就是为什么会出现断流的情况，表现为网页打不开，游戏玩不了，qq能收发信息，好奇怪，我什么插件都还没开的，为啥呢

发自我的iPhone

DHDAXCW, 2021-02-09:
刚刷上之后时间不准，同步一下时间（系统→系统→时间同步一下）或者重启联网就可以自动同步时间后就好了

DHDAXCW, 2021-02-09:
> 大佬，新版已经用上，就是为什么会出现断流的情况，表现为网页打不开，游戏玩不了，qq能收发信息，好奇怪，我什么插件都还没开的，为啥呢 发自我的iPhone
> […](#)

刚刷上之后时间不准，同步一下时间（系统→系统→时间同步一下）或者重启联网就可以自动同步时间后就好了

ljtao020, 2021-02-09:
哎，还是不行，这个现象我在其它估计也遇到过，现在就只能qq收发信息，网页打不开，游戏的链接有不是完全断线，显示服务器连接超时

发自我的iPhone

DHDAXCW, 2021-02-09:
把所有设备都重启一下试试。我刚才试了我的没有问题

ljtao020, 2021-02-09:
好的，这个会和sd卡有关系吗，还是我这个r4s本来就有问题呢？

发自我的iPhone

DHDAXCW, 2021-02-09:
不会。要不把你重刷吧。先刷r2s再刷r4s

ljtao020, 2021-02-09:
好的，刷完之后需要回复设置吗？

发自我的iPhone

------------------ 原始邮件 ------------------
发件人: 骷髅头 <notifications@github.com&gt;
发送时间: 2021年2月9日 22:07
收件人: DHDAXCW/NanoPi-R4S-2021 <NanoPi-R4S-2021@noreply.github.com&gt;
抄送: ljtao020 <61601904@qq.com&gt;, Author <author@noreply.github.com&gt;
主题: 回复：[DHDAXCW/NanoPi-R4S-2021] 温度没显示 (#2)

DHDAXCW, 2021-02-09:
当然不行，需要重新设置，恢复备份，容易让系统崩溃

ljtao020, 2021-02-11:
大佬，新年好，之前那个问题解决了，是因为光猫松了，收光功率都去到30多了，不过还有个问题，现在R4s大概每天晚上4点就会自动关机了，为什么呢，早上起来一个灯都没亮，好像之前的固件也会这样的

发自我的iPhone

DHDAXCW, 2021-03-19:
祝你好运

closed 2021-03-19 by DHDAXCW