DHDAXCW/NanoPi-R4S-R4SE #6: 增加网卡驱动

DHDAXCW, 2021-02-11:
你们有需要的wifi驱动，可以留言，留下网卡信息，我后面看到就装固件里

Evens3body, 2021-02-12:
请教为何我刷如了2.11版本后，插上无线网卡依然不亮。是不是需要在openwrt系统内部一通操作才行？

DHDAXCW, 2021-02-12:
无线芯片是什么？

Evens3body, 2021-02-12:
> 无线芯片是什么？
就是那天我提供的这个
Realtek 8812BU Wireless

Evens3body, 2021-02-12:
目前插上灯并不会亮

DHDAXCW, 2021-02-12:
openwrt能看到无线吗？

Evens3body, 2021-02-12:
> openwrt能看到无线吗？

就是不能啦

Evens3body, 2021-02-12:
我确定这个无线网卡肯定是好的，因为我在PC，windows环境下试过没问题！

DHDAXCW, 2021-02-12:
下午我再换个驱动吧🤕我这也能用8812bu  就是缺中国码。

Evens3body, 2021-02-12:
我在r2s上刷过klever 1988 的版本，插上无线网卡可以驱动，但测试几次确实不太稳定，不知道是不是网卡兼容性问题。

Evens3body, 2021-02-12:
> 下午我再换个驱动吧🤕我这也能用8812bu 就是缺中国码。

非常感谢，我也想学学编译，如果有入门的教程，方便的话可以推荐一下。

DHDAXCW, 2021-02-12:
今天加了这个驱动rtl88x2bu，先试试。我这个瞎学的，属于杂交编译。你先学会Linux基础看就懂了

Evens3body, 2021-02-13:
> 今天加了这个驱动rtl88x2bu，先试试。我这个瞎学的，属于杂交编译。你先学会Linux基础看就懂了

请问具体下载哪个？

DHDAXCW, 2021-02-13:
没有，还没编译出来，等6小时看看再说，昨天编译失败了

Evens3body, 2021-02-13:
> 没有，还没编译出来，等6小时看看再说，昨天编译失败了

辛苦了，过年好

Roger-1987, 2021-03-02:
大佬，我新买的8812BU芯片的USB3.0的无线网卡，固件是你的3月2日最新版，插上网卡后能马上识别，但是目前有两个问题：
（1）修改WiFi设置必须重启路由才能生效，否则连不上WiFi，显示密码不正确，重启后就正常；
（2）5GWiFi下速度只有22-25Mbps，完全达不到油管上一个UP主的250-300+Mbps的速度，USB2.0的网卡都比这个高吧，不知道是不是驱动兼容性的问题，切换信道什么的方法我都试过了，都是这个速度。
最后感谢大佬的无私奉献，固件很好用，就是后台更改设置显示有点慢，不跟X86架构比，比树莓派4B+也要慢得多，不过完全不影响正常使用。

closed 2021-03-19 by DHDAXCW

DHDAXCW, 2021-11-02:
> netgear A6210

不是默认带上了吗

gnastnosaj, 2021-12-19:
大佬，能否帮忙加下TL-WDN7201H驱动，RTL8814AU芯片
[TL-WDN7201H V1.0 Linux驱动程序.zip](https://github.com/DHDAXCW/NanoPi-R4S/files/7740534/TL-WDN7201H.V1.0.Linux.zip)
H

zhangguanzhang, 2022-06-24:
rtl88x2bu 在哪里，我找到了个 不知道能用不
```
    git clone --depth=1 --single-branch openwrt-21.02 https://github.com/erintera/openwrt-rtl8812bu-package.git package/kernel/rtl88x2bu
    echo 'CONFIG_PACKAGE_kmod-88x2bu=y' >> .config
```