DHDAXCW/NanoPi-R4S-R4SE #9: 更新后新功能菜单上出不来

derekling, 2021-02-14:
后台直接上传的新版固件更新，重启后看不到新功能，怎么解决

DHDAXCW, 2021-02-14:
不要用此方法操作，直接卡刷。

cuikai-wh, 2021-02-14:
> 不要用此方法操作，直接卡刷。

开发者你好，想请教一下2个问题哦：

1. 请问为什么不能用上传固件更新呢？这个功能还挺方便的，因为不需要把卡拔出来。期待后续可以得到支持哦
2. 假设必须要直接卡刷，看到你主页的提示，需要先刷R2S，再刷R4S即可清除卡里之前一个版本的数据。请问为什么要先刷2s呀？看到网上的教程是，直接覆盖即可实现清除。

DHDAXCW, 2021-02-14:
1 上传固件，因为我改了我自己补丁，uboot就不一样的，另外lean源码本身就存在这个问题，这样的话，哪天更新了插件，你恢复备份，容易导致某个插件崩溃。所以这就是我不建议这样做，直接卡刷
2 因为sq格式，刷上后数据依然还在，特别是我从ttly输入命令加上温度，刷新版（不存在温度）结果显示。。有个小分区没有清除掉。这格式只能读不能写，所以没办法覆盖数据。只有先刷r2s把数据覆盖再重刷r4s即可。或者删除分区格式化再刷都是可以的。
目前这两点，我这一个月刷了好多次才知道。现在我用第二种方法来刷

DHDAXCW, 2021-02-14:
> > 不要用此方法操作，直接卡刷。
> 
> 开发者你好，想请教一下2个问题哦：
> 
> 1. 请问为什么不能用上传固件更新呢？这个功能还挺方便的，因为不需要把卡拔出来。期待后续可以得到支持哦
> 2. 假设必须要直接卡刷，看到你主人的提示，需要先刷R2S，再刷R4S可以清除卡里之前一个版本的数据。请问为什么要先刷2s呀？看到网上的教程是，直接覆盖即可实现清除。

1 上传固件，因为我改了我自己补丁，uboot就不一样的，另外lean源码本身就存在这个问题，这样的话，哪天更新了插件，你恢复备份，容易导致某个插件崩溃。所以这就是我不建议这样做，直接卡刷
2 因为sq格式，刷上后数据依然还在，特别是我从ttly输入命令加上温度，刷新版（不存在温度）结果显示。。有个小分区没有清除掉。这格式只能读不能写，所以没办法覆盖数据。只有先刷r2s把数据覆盖再重刷r4s即可。或者删除分区格式化再刷都是可以的。
目前这两点，我这一个月刷了好多次才知道。现在我用第二种方法来刷

cuikai-wh, 2021-02-14:
> 1 上传固件，因为我改了我自己补丁，uboot就不一样的，另外lean源码本身就存在这个问题，这样的话，哪天更新了插件，你恢复备份，容易导致某个插件崩溃。所以这就是我不建议这样做，直接卡刷
> 2 [...]

收到了，感谢回复！ 还有2个小白问题想问一下，希望不要嫌弃哈：
1. 请问格式化的意思是，直接格式化掉吗，还是需要把分区什么的弄一下？以及，格式化后建议用什么文件系统呢？是exFAT吗？
2. 重刷固件之后，请问你建议是重新配置一遍，还是可以支持上传之前备份的配置呢？

DHDAXCW, 2021-02-14:
> 收到了，感谢回复！ 还有2个小白问题想问一下，希望不要嫌弃哈：
>
> 1. 请问格式化的意思是，直接格式化掉吗，还是需要把分区什么的弄一下？以及，格式化后建议用什么文件系统呢？是exFAT吗？
> 2. 重刷固件之后，请问你建议是重新配置一遍，还是可以支持上传之前备份的配置呢？

1 软件叫DiskGenius 硬盘分区 选择你TF卡删除所有分区，再建立exFAT格式分区，格式化后即可干净
2刷固件当然需要手动配置一遍，千万别用恢复备份，容易导致某个插件崩溃

cuikai-wh, 2021-02-14:
收到了你的建议，十分感谢！

derekling, 2021-02-15:
按这个说法，每次升级都折腾一次要累死了，有米有什么更方便的无损升级办法呢

DHDAXCW, 2021-02-15:
> 按这个说法，每次升级都折腾一次要累死了，有米有什么更方便的无损升级办法呢

哈哈 我很难嘛。我试过sq格式升级，但是感觉像没有升级一样。之后我就用ext4刷上系统后再用ext4包升级，有变化的。我是今天刚刚试了。你可以试试，但不能保证某个插件能正常用，，，，只要固件稳定了，就别再刷了嘛

derekling, 2021-02-18:
> > 按这个说法，每次升级都折腾一次要累死了，有米有什么更方便的无损升级办法呢
> 
> 哈哈 我很难嘛。我试过sq格式升级，但是感觉像没有升级一样。之后我就用ext4刷上系统后再用ext4包升级，有变化的。我是今天刚刚试了。你可以试试，但不能保证某个插件能正常用，，，，只要固件稳定了，就别再刷了嘛

是用ext4固件就能直接后台上传升级吗？和sq格式功能上有啥区别？

derekling, 2021-02-18:
另外想问问和你另外一个编译的friendwrt的固件相比有啥区别，怎么选择呢？

closed 2021-03-19 by DHDAXCW