DHDAXCW/NanoPi-R4S-R4SE #11: 2.15固件报错

hteane, 2021-02-15:
目前发现Passwall添加Xray分流规则会出现以下报错“/usr/lib/lua/luci/dispatcher.lua:469: Failed to execute cbi dispatcher target for entry '/admin/services/passwall/shunt_rules'.
The called action terminated with an exception:
/usr/lib/lua/luci/cbi.lua:900: table index is nil
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/dispatcher.lua:469: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:121: in function </usr/lib/lua/luci/dispatcher.lua:120>”

DockerMan容器新建解析命令行点提交出现“/usr/lib/lua/luci/dispatcher.lua:469: Failed to execute cbi dispatcher target for entry '/admin/services/docker/newcontainer/DOCKERCLI -d --name jellyfin --user uid:gid --net=host --volume /path/to/config:/config --volume /path/to/cache:/cache --mount type=bind,source=/path/to/media,target=/media --restart=unless-stopped jellyfin/jellyfin'.
The called action terminated with an exception:
...sr/lib/lua/luci/model/cbi/dockerman/newcontainer.lua:111: attempt to concatenate a nil value
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/dispatcher.lua:469: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:121: in function </usr/lib/lua/luci/dispatcher.lua:120>“

请问怎么解决啊？

DHDAXCW, 2021-02-15:
1 passwall那边在修了，等新版本吧。我今天弄下passwall  明天出16版看看吧。可以用ssrp或者hellowrold
  Docker那个我不知道了。今晚换个源码试试吧，我也没玩过这个，不好意思哈
2 顺便问下，你怎么刷机的？有没有卡刷？有没有用恢复备份？

> /usr/lib/lua/luci/dispatcher.lua:469：无法为条目“ / admin / services / passwall / shunt_rules”执行cbi调度程序目标。
> 调用的操作终止，但有以下异常：
> /usr/lib/lua/luci/cbi.lua : 900：表索引为nil
> 堆栈回溯：
> [C]：在函数“声明”
> / usr / lib / lua / luci / dispatcher中。 lua：469：在'dispatch'
> 函数中/usr/lib/lua/luci/dispatcher.lua:121：在函数</usr/lib/lua/luci/dispatcher.lua:120>”
> 
> DockerMan容器新建解析命令行点提交出现“ /usr/lib/lua/luci/dispatcher.lua:469：无法执行条目'/ admin / services / docker / newcontainer / DOCKERCLI -d --name jellyfin的cbi调度程序目标--user uid：gid --net = host --volume / path / to / config：/ config --volume / path / to / cache：/ cache --mount type = bind，source = / path / to / media ，target = / media --restart =除非停止jellyfin / jellyfin'。
> 调用的操作终止，但发生以下异常：
> ... sr / lib / lua / luci / model / cbi / dockerman / newcontainer.lua：111：尝试串联nil值
> 堆栈回溯：
> [C]：在函数'assert'中
> / usr / lib / lua / luci / dispatcher.lua：469：在函数'dispatch'中
> /usr/lib/lua/luci/dispatcher.lua:121：在函数</usr/lib/lua/luci/dispatcher.lua中： 120>“
> 
> 请问怎么解决啊？

1 passwall那边在修了，等新版本吧。我今天弄下passwall  明天出16版看看吧。可以用ssrp或者hellowrold
  Docker那个我不知道了。
2 顺便问下，你怎么刷机的？有没有卡刷？有没有用恢复备份？

DHDAXCW, 2021-02-15:
> /usr/lib/lua/luci/dispatcher.lua:469：无法为条目“ / admin / services / passwall / shunt_rules”执行cbi调度程序目标。
> 调用的操作终止，但有以下异常：
> /usr/lib/lua/luci/cbi.lua : 900：表索引为nil
> 堆栈回溯：
> [C]：在函数“声明”
> / usr / lib / lua / [...]

1 passwall那边在修了，等新版本吧。我今天弄下passwall  明天出16版看看吧。可以用ssrp或者hellowrold
    Docker那个我不知道了。
2 顺便问下，你怎么刷机的？有没有卡刷？有没有用恢复备份？

hteane, 2021-02-15:
严格卡刷，甚至我觉得用刷R2S固件再刷回来以及flash中途拔卡两个都感觉清除得不彻底我干脆还换了张卡。Passwall那个我解决了，可能是添加的时候名字不能包含某些字符，我创建用DIsney+就不可以，后来用DISNEY就可以进入编辑页面然后保存后又报错一次但是返回页面是保存了规则的。Docker那个我也试了下，就官方Jellyfin的代码不行，我用QBT QEE的代码不报错，不知道具体什么原因。

hteane, 2021-02-15:
关于恢复，我自从听了你说的就再也没有导出过备份了。。

DHDAXCW, 2021-02-15:
我的卡写坏了....每天都要刷10次多😭  你说那个容器，方便提供源码吗？

hteane, 2021-02-15:
还有个问题想请教下，每次刷完要配置好久，DDNS，Passwall负载均衡和分流，softether什么的，还要扩容overlay和/opt/docker，好花时间啊。因为我是新手小白，不清楚比如新包里的passwall更新了之后，我有办法可以不刷新固件单独更新我想要的应用吗？因为我今天以为是2.15固件的问题，我又刷回2.11固件了，问题是一样的，但我发现好像2.11固件里的passwall的V2ray也没有了，好像是最新版（不确定，我后来又刷回2.15了没在2.11仔细看），所以问一下有没有这个可能。。。

DHDAXCW, 2021-02-15:
v2ray已加了，可以在公告看。明天就用上了。不过你有v2ray也能用，只是xray代v2ray而已，不影响

hteane, 2021-02-15:
docker hub是这个：https://hub.docker.com/r/jellyfin/jellyfin
然后我是在下面描述里去他们官网找到的docker命令行如下：
Using Docker command line interface:

docker run -d \
 --name jellyfin \
 --user uid:gid \
 --net=host \
 --volume /path/to/config:/config \
 --volume /path/to/cache:/cache \
 --mount type=bind,source=/path/to/media,target=/media \
 --restart=unless-stopped \
 jellyfin/jellyfin

DHDAXCW, 2021-02-15:
> docker hub是这个：https://hub.docker.com/r/jellyfin/jellyfin
> 然后我是在下面描述里去他们官网找到的docker命令行如下：
> Using Docker command line interface:
>
> docker run -d
> --name jellyfin
> --user uid:gid
> --net=host
> --volume [...]

谢谢  我研究一下哈

hteane, 2021-02-15:
对的是不影响，只是这个让我恍惚以为卡没刷彻底，我同时执行了先刷R2S再刷回来，然后第二次刷的时候拔卡，然后我想，如果不没刷彻底的原因，那就是passwall的软件包是联网更新了，不知带是不是这样，如果是的话，那以后可以不用追新固件了，我觉得我R4S到手后从2.11固件开始刷，每个都还挺稳定了至少在我需要的范围内，如果可以单独更新passwall、dockerman什么的真是谢天谢地了

DHDAXCW, 2021-02-15:
passwall单独更新很麻烦，这是集成插件。还不如编译就行了。用稳定不卡就行。不过你卡刷了后，数据依然没变化，可以按系统→恢复/备份→重置之后就变成那样

hteane, 2021-02-15:
不是很懂哎，哈哈，还有好多需要学习，我只知道不刷卡重置之后会系统overlay会重新挂载到原始盘位，我只需要再重新挂载我扩容的分区到overlay可以恢复配置，但是重新刷卡之后，我扩容overlay新建的分区就没了，所有的配置还是得重新来一遍包括扩容～

DHDAXCW, 2021-02-15:
对呀。要不你刷一个稳定就别刷了。哪天固件不稳定再刷也行。天天刷折磨人。我呢，天天刷都是测试固件的😅

hteane, 2021-02-15:
另外2.15设置AdGuard的时候出现Error！bin is a dir什么之类的报错没办法保存应用，后来自己乱搞通过r m -rf /usr/bin/AdGuardHome把这个文件夹的内容全删掉再重新mkdir才行的，不知道什么原因，但是我第一次刷2.15的时候好像没碰到，第二次碰到的。。。

hteane, 2021-02-15:
好吧，如果没有重大更新或者或者有重要的功能不能用我就不刷了。。。哈哈哈，我刷机没有卸载移动硬盘导致现在移动硬盘数据都丢了，等朋友带windows来拯救我，我硬盘ntfs格式用mac没办法修复。。。哎

hteane, 2021-02-15:
对了，最后一个问题，我为了软路由才注册的GitHub一时半会真的不是很懂这个UI，然后想问下你每次新的release之后会删掉之前的吗，我看每次只有一两个最新的release，没办法选择下载中间某个时间点的固件版本对吧

DHDAXCW, 2021-02-15:
我去，这也可以把硬盘数据丢了？？😧涨知识了。。AdGuard那个我一直在用。貌似没有出现你这种情况。不过放心。我会继续搞完整的。
关于那个rerlase我已改了7天删除。就是保证固件正常稳定。不稳定我就删除

DHDAXCW, 2021-02-15:
每次出新固件，就自动把前7天固件删除。从yml代码倒数第二行release删除，定期7天保留。

hteane, 2021-02-15:
所以以后特别稳定的版本还是会保留发布的对吧？我就是太爱折腾，这几天都刷十好几次了，然后硬盘茶软路由上的，每次拔卡前直接断电，搞了几次发现硬盘数据没了，就已使用空间还在，但文件消失了，应该用windows还是可以恢复的

DHDAXCW, 2021-02-15:
只删除近7天固件，都是自动删除的

hteane, 2021-02-15:
又有新问题，Aria2的配置文件到底在哪里，我指定的配置文件夹位置里只有aria2.conf.main的自动生成的文件，我修改覆盖路由器上的但是没办法从它启动和应用，WEBUI设置的话又会直接覆盖，死循环。。。WEBUI里批量添加Tracker简直要死了。。。

DHDAXCW, 2021-02-15:
这个昨天就找到了，等今天的吧

hteane, 2021-02-16:
自己搞定了，在../erc/config下面有个aria2的文件，这个就是配置文件，如果是想批量添加tracker的话甚至不需要手动改配置文件，在WEBUI里找到添加额外tracker然后在一行里直接复制用半角逗号分隔的多个tracker保存应用会自动识别然后分条。很方便！

closed 2021-03-19 by DHDAXCW