DHDAXCW/NanoPi-R4S-R4SE #13: #关于软件包扩容问题建议

DHDAXCW, 2021-02-19:
目前软件包容量是200MB多，占用22%，剩下那么多，不知道干嘛用的。可以扩容到1000MB之后就可以支持在线升级。但是，这样做，解压包就大了1000MB，，，很愁，给个建议 要不要搞？

derekling, 2021-02-19:
当然要搞 方便多了

gj0518, 2021-02-19:
非常建议搞一个, 卡那么大不用也浪费, 强烈要求大佬搞一个, 省得扩容了

1090549930, 2021-02-19:
搞

hteane, 2021-02-19:
搞！我overlay自己扩容都扩了10g了。。现在16g卡这么便宜，宽带至少百兆了吧，下载包大点都无所谓啦，这点下载时间比起来重新配置的时间来说不值一提。

derekling, 2021-02-20:
楼主加油啊

cyy911, 2021-02-20:
@hteane 非常同意你说的话

DHDAXCW, 2021-02-21:
目前在测试在线升级。就是内部解压出错问题，明天尝试gz升级，今天很累了。对了还有一件事，就是 要给软件包扩容多大空间，在这里提出来，根据你们要的容量比例一下就决定选择多大容量

derekling, 2021-02-21:
> 目前在测试在线升级。就是内部解压出错问题，明天尝试gz升级，今天很累了。对了还有一件事，就是 要给软件包扩容多大空间，在这里提出来，根据你们要的容量比例一下就决定选择多大容量

64或者128g tf卡应该是主流

1090549930, 2021-02-21:
啊这

1090549930, 2021-02-21:
16g的...

DHDAXCW, 2021-02-21:
尴尬了。。。。可能为16g考虑了  软件包扩容2g够用，写卡时间变长了，合着用呗

cyy911, 2021-02-21:
同 16G

hteane, 2021-02-21:
个人128g，不过觉得16g应该是主流，扩到3g一般够用并且16g卡也无压力，剩下空间也可以自行扩docker。个人建议。

ljtao020, 2021-02-22:
真能在线升级，也不用老写卡啊

DHDAXCW, 2021-02-22:
软件包扩容问题，在线升级问题已搞定了，我一会儿放主页下面升级说明。。😀可是这包会很大的哦

DHDAXCW, 2021-02-22:
今天的版本扩容问题改到明天23号。如果你们软件包不够的话，等明天再次卡刷，就能正常使用在线升级了

hteane, 2021-02-22:
太棒了！就是问一下，自己扩容overlay的可以直接luci升级吗？升完之后挂载的overlay是系统新置顶的位置，我原来overlay的文件还在的话会不会刷不彻底？

DHDAXCW, 2021-02-22:
> 太棒了！就是问一下，自己扩容overlay的可以直接luci升级吗？升完之后挂载的overlay是系统新置顶的位置，我原来overlay的文件还在的话会不会刷不彻底？

这个我没有试过

1090549930, 2021-02-22:
> 软件包扩容问题，在线升级问题已搞定了，我一会儿放主页下面升级说明。。😀可是这包会很大的哦

下载慢可以用镜像站嘛，
https://github.91chifun.workers.dev//https://github.com/DHDAXCW/NanoPi-R4S-2021/releases/
其他镜像站都行吧，至少比github快

DHDAXCW, 2021-02-22:
> > 软件包扩容问题，在线升级问题已搞定了，我一会儿放主页下面升级说明。。😀可是这包会很大的哦
> 
> 下载慢可以用镜像站嘛，
> https://github.91chifun.workers.dev//https://github.com/DHDAXCW/NanoPi-R4S-2021/releases/
> 其他镜像站都行吧，至少比github快

好主意啊，谢谢提供！我这就去弄

cyy911, 2021-02-22:
> > 软件包扩容问题，在线升级问题已搞定了，我一会儿放主页下面升级说明。。😀可是这包会很大的哦
> 
> 下载慢可以用镜像站嘛，
> [github.91chifun.workers.dev//https://github.com/DHDAXCW/NanoPi-R4S-2021/releases](https://github.91chifun.workers.dev//https://github.com/DHDAXCW/NanoPi-R4S-2021/releases/)
> 其他镜像站都行吧，至少比github快

我觉得都用 R4S 了，加速下载的能力还是有的。

derekling, 2021-02-22:
怎么扩容 ，给个0基础傻瓜教程吧楼主，拜谢

hteane, 2021-02-23:
> 怎么扩容 ，给个0基础傻瓜教程吧楼主，拜谢

不用扩容啦，这不大佬已经安排上了吗。而且扩容overlay只适用于sq文件系统，这个在线升级好像只支持ext4我看说明里说...

derekling, 2021-02-24:
楼主，我看软件包大小没明显变化，到底刷哪个能实现在线升级呀

DHDAXCW, 2021-02-24:
在线升级有问题，暂时先取消。后期处理一下

lingzhuxin, 2021-02-24:
> 在线升级有问题，暂时先取消。后期处理一下

我还特意从18为了在线升级上来的，然后就刷了一天…

DHDAXCW, 2021-02-24:
在线升级有问题。暂停用在线更新😭我研究研究怎么解压更新

lingzhuxin, 2021-02-25:
> 在线升级有问题。暂停用在线更新😭我研究研究怎么解压更新
所以这个25版的还是得卡刷上来吗？我歇歇，不能刷了

choz333, 2021-02-25:
首先辛苦大佬了~~
我是当主路由用的，每次都要卡刷太苦了，拨卡，拨机器，升级完所有东西都要重新设置 ，等待大佬的固件能在LUCI面更新
，加油！

Jason-Rock, 2021-02-26:
搞起，加油大佬

cscecyhf, 2021-02-27:
可以电脑上解压，直接上传解压后得文件么？

lingzhuxin, 2021-02-27:
大佬现在更新没日志了我好慌啊，等你在线更新上线中……

DHDAXCW, 2021-02-27:
> 大佬现在更新没日志了我好慌啊，等你在线更新上线中……

一直都在优化，没有啥可写的，等搞定了再写

lingzhuxin, 2021-03-02:
那我感觉完全可以一直拖到你整上在线升级我再刷了，先这么稳着用吧

closed 2021-03-19 by DHDAXCW

SyllableSpec, 2021-03-25:
>目前，软件包容量超过200MB，占22%，剩下那么多，不知道干嘛用。可以扩容到1000MB，可以支持在线升级。但是，这样做，解压包会大1000MB，，，很愁，给个建议，好吗？

搞！