DHDAXCW/NanoPi-R4S-R4SE #14: 端口转发貌似不起作用（公网ip）

1090549930, 2021-02-19:
也可能是我操作原因（？）

DHDAXCW, 2021-02-19:
我用的没问题

1090549930, 2021-02-19:
10000和20000端口有占用吗我用的这2个

1090549930, 2021-02-19:
我试了upnp也不行

1090549930, 2021-02-19:
[image]
这里有问题吗，求教

1090549930, 2021-02-19:
[image]
这里有问题吗，求教

DHDAXCW, 2021-02-19:
可能是，明天我查查问题吧

1090549930, 2021-02-19:
谢谢了a

DHDAXCW, 2021-02-19:
用iptables转发，你的 iptables没放行吧

DHDAXCW, 2021-02-19:
[image]

DHDAXCW, 2021-02-19:
[image]

DHDAXCW, 2021-02-19:
[image]
[image]

1090549930, 2021-02-19:
iptables转发没用过，，，以前都是在防火墙里面添加转发

xbclub, 2021-02-20:
能否提供详细的转发步骤，与验证方式和结果

1090549930, 2021-02-20:
> 能否提供详细的转发步骤，与验证方式和结果

[image]
哪有什么步骤的，像这样，以前我也都是怎么转发的，求教谢谢
另外没用过iptables..

hongye1990, 2021-02-20:
我的也有点问题，虽然可以转发，但是只能从其他的外部网络（比如说手机热点）访问到被转发的端口，如果直接从内网访问是不可以的。

1090549930, 2021-02-20:
> 我的也有点问题，虽然可以转发，但是只能从其他的外部网络（比如说手机热点）访问到被转发的端口，如果直接从内网访问是不可以的。

还真是，我这边用流量也行

xbclub, 2021-02-20:
traceroute 或者tracert 看下路由

closed 2021-03-19 by DHDAXCW