DHDAXCW/NanoPi-R4S-R4SE #20: 卡刷24版打不开，卡刷23版进来后adh死活更新不了

lingzhuxin, 2021-02-24:
自己换了一个官网的https://github.com/AdguardTeam/AdGuardHome/releases/download/v0.105.1/AdGuardHome_linux_mips_softfloat.tar.gz这个链接进去更新也不行，一直都是显示没有核心，更新完了最新的这个版本也是不会显示，看网上说可以自己去更新，用putty代码更新完了再回来还是显示没有……我裂开

DHDAXCW, 2021-02-24:
最近源码都没改，不会出错的。设置里保留核心和原来的配置，就不用手动更新核心，即可解决每次更新固件之后，手动更新core的问题。用ext4格式

lingzhuxin, 2021-02-24:
> 最近源码都没改，不会出错的。设置里保留核心和原来的配置，就不用手动更新核心，即可解决每次更新固件之后，手动更新core的问题。用ext4格式

唔，问题就在这里了以前一直用的是那个s格式的，然后设置里我也开了那个升级保存核心和配置，但是今天看到说ext4的才可以在线升级我就又重新格式化了，然后卡刷上了23版，就是不知道为什么会什么都设置好了，进adh就怎么更新都检测不了我有内核

lingzhuxin, 2021-02-24:
我刚才还试了是不是我构架选错了，然后用https://github.com/AdguardTeam/AdGuardHome/releases/download/v0.105.1/AdGuardHome_linux_arm64.tar.gz更新链接去代替，怎么点都能下载成已更新，但是底下的显示就是一直core error，我实在是黔驴技穷了

DHDAXCW, 2021-02-24:
对了，顺便问下，24版启动不了？

lingzhuxin, 2021-02-24:
> 对了，顺便问下，24版启动不了？

对，我首先刷的就是24版的，s文件和e文件我都试了，我一开始还以为是我换版本的问题，我又用u盘工具初始了卡，再刷n遍都是输入了密码后就是几行代码……我忍不了，就刷了23版，就好了……我觉得一会儿我还得一刷解千愁，可之前保存了adh的文件也架不住我这么初始化啊，实在是想不到为什么更新不了，备份还原不能用我也试了不还原更，一天就这么刷没了

lingzhuxin, 2021-02-24:
你要再更一版我就不刷23了，里外都是要卡刷的，我再刷新版好了

DHDAXCW, 2021-02-24:
不好意思啊，我今天就看看源码咋了。。。再不行我就炸了它

lingzhuxin, 2021-02-24:
加油！我反正是纯小白……实在是我能力范围内的谷歌能解决的办法我都试了，剩下的就看大佬你们操作了

lingzhuxin, 2021-02-24:
> 最近源码都没改，不会出错的。设置里保留核心和原来的配置，就不用手动更新核心，即可解决每次更新固件之后，手动更新core的问题。用ext4格式

不过我还是很好奇，每次更新我都是卡刷的，工具直接就提示我删除分区了，那我更新了后设置的保存核心和配置还有用？

closed 2021-03-19 by DHDAXCW