DHDAXCW/NanoPi-R4S-R4SE #23: xray无法fullcone

nuobeike, 2021-02-28:
xray1.3之后支持fullcone，游戏加速非常爽，但是这款固件怎么折腾都是udpblock，到底是哪里出了问题呢@@，别的固件可以fullcone但是速度没你的快，就想用你的高大全，求帮忙看看呀

DHDAXCW, 2021-02-28:
已知，明天刷吧。我今晚就整理顺序一下

DHDAXCW, 2021-02-28:
我记得应该是docker导致的，你把容器卸载再重启试试，如果不行明天看看吧，我只知道那个容器导致的。顺便提供一下你说别的固件？

nuobeike, 2021-02-28:
1988的https://github.com/klever1988
洋葱的https://drive.google.com/file/d/1YBM64n_8FGgR745AXoBRzTWePDMdcFgo/view?usp=sharing

DHDAXCW, 2021-03-01:
今天xray-plugin删除了。明天看看

DHDAXCW, 2021-03-01:
3.1版没有这个xray-plugin

xbclub, 2021-03-03:
docker导致的问题，卸载是没有用的，一旦编译进去udp代理就gg.

DHDAXCW, 2021-03-03:
> docker导致的问题，卸载是没有用的，一旦编译进去udp代理就gg.

确实有问题，但是我打掉，下不了手。。。有人在用。。。我还能咋办，过几天我找解决方案吧

hteane, 2021-03-04:
> > docker导致的问题，卸载是没有用的，一旦编译进去udp代理就gg.
> 
> 确实有问题，但是我打掉，下不了手。。。有人在用。。。我还能咋办，过几天我找解决方案吧

啊，docker我在用。。。。。我还想问是用的哪个插件实现游戏加速的，我openclash 加速switch各种模式试下来NAT能到A但是无法联机，线路是v2ray的，虽然买了UU但是真心不想花两份钱

nuobeike, 2021-03-06:
能否先出个不带docker的版本临时规避下哇~

closed 2021-03-19 by DHDAXCW