DHDAXCW/NanoPi-R4S-R4SE #26: R4S 1G用不了后面的固件

Neilson8, 2021-03-03:
我能刷2月6号的固件，后面3月的固件都试了全部用不了，怎么回事？我这有三张内存卡，不同品牌的，全部用不了，为什么2月6号的固件就能用，后面的用不了，刷好了装路由里，系统 灯不亮，很多人1G内存的也是用 不了3月的固件

DHDAXCW, 2021-03-03:
好多人这么夸张？我蹲群看那人刷最新版3号，也是1g，能正常启动。推荐用rufs写固件。用ext4刷，不愿意用ext4，那就先低格再刷sq也行

Neilson8, 2021-03-03:
> rufs写固件

搞定了，原来是刷卡工具问题

Neilson8, 2021-03-03:
> 好多人这么夸张？我蹲群看那人刷最新版3号，也是1g，能正常启动。推荐用rufs写固件。用ext4刷，不愿意用ext4，那就先低格再刷sq也行

3月3号的固件有USB网卡驱动 ？怎么用，插了没反应，我是RTL8818CE

DHDAXCW, 2021-03-03:
明天我给你适配👌

Neilson8, 2021-03-03:
> 明天我给你适配👌

先谢了，型号是RTL 8818EU USB 2.0

Neilson8, 2021-03-04:
> 明天我给你适配👌

好了吗，是不是下载你今天放出的固件来刷？

TommyMyCat, 2021-03-11:
请问骷髅头大神，我是一个新手小白，第一次使用软路由，刚刚购买了R4S-1G，只是用来科学一下，没有其它的需求。

请问刷 FriendlyWRT-R4S 还是 NanoPi-R4S 呢？
如果后者的话，nanopi-r4s-ext4-sysupgrade.img.gz
，nanopi-r4s-rootfs.tar.gz
，nanopi-r4s-squashfs-sysupgrade.img.gz
该如何选择呢？

希望得到您的帮助。

closed 2021-03-19 by DHDAXCW