DHDAXCW/NanoPi-R4S-R4SE #27: 可以增加qbittorrent吗

Neilson8, 2021-03-03:
ARIA2和TRAMISSION很多种子下不动，迅雷就能下，听说qbittorrent下载种子也很给力，可以加吗

lingzhuxin, 2021-03-03:
吸血雷能下那是因为它吸血……

DHDAXCW, 2021-03-03:
有源码链接吗？这个必须搞，方便把源码留在这，明天我开搞，后天才能用

Neilson8, 2021-03-03:
> 有源码链接吗？这个必须搞，方便把源码留在这，明天我开搞，后天才能用

https://www.right.com.cn/FORUM/thread-1456090-1-1.html 我昨天 找了很多IPK，想安装 ，找不到，在恩山论坛看到相关资料，你去看下

Neilson8, 2021-03-03:
> 有源码链接吗？这个必须搞，方便把源码留在这，明天我开搞，后天才能用

有办法增加小米路由器那个迅雷下载插件吗，那个很好用，跟迅雷差不多，支持很多格式，种子都下得动

hteane, 2021-03-03:
插一句，qbt在docker上运行就行，个人觉得设置挺方便的。

DHDAXCW, 2021-03-03:
> 插一句，qbt在docker上运行就行，个人觉得设置挺方便的。

有教程吗？

hteane, 2021-03-04:
先运行命令“docker pull johngong/qbittorrent:qee-latest”下载镜像，这个是集成额外tracker的版本。然后去容器里新建容器-输入命令：
 docker create  \
    --name=qbittorrent  \
    -e WEBUIPORT=8989  \
    -e UMASK=022  \       
    -e UID=65534  \（65534这个是nobody用户ID，权限最低，我个人这么选的，如果用户和组选错可能无法读写和执行下下来的文件，你也可以自己选，下面GID同理）
    -e GID=65534  \  
    -p 6881:6881  \
    -p 6881:6881/udp  \
    -p 8989:8989  \
    -v /opt/docker/qbt/config:/config  \
    -v /mnt/sda1/Downloads（冒号前的路径是下载文件保存路径，自己改）:/Downloads  \
    --restart unless-stopped  \
   johngong/qbittorrent:qee-latest

网络选择host，然后启动去路由地址:8989的WEBUI管理就行，初始帐号admin，密码adminadmim

Neilson8, 2021-03-04:
> 插一句，qbt在docker上运行就行，个人觉得设置挺方便的。

你看上面恩山的帖子，有IPK包的，IPK肯定比容量容易操作吧

hteane, 2021-03-04:
> > 插一句，qbt在docker上运行就行，个人觉得设置挺方便的。
> 
> 你看上面恩山的帖子，有IPK包的，IPK肯定比容量容易操作吧

并不觉得，我上面的回复就是所有你需要的东西，照着做2分钟就搞定了。你那个帖子看完都不知2分钟。而且有官方源和第三方反吸血的版本自己可以选，随时更新，你选官方版本得添加额外tracker。

Neilson8, 2021-03-04:
> > > 插一句，qbt在docker上运行就行，个人觉得设置挺方便的。
> > 
> > 
> > 你看上面恩山的帖子，有IPK包的，IPK肯定比容量容易操作吧
> 
> 并不觉得，我上面的回复就是所有你需要的东西，照着做2分钟就搞定了。你那个帖子看完都不知2分钟。而且有官方源和第三方反吸血的版本自己可以选，随时更新，你选官方版本得添加额外tracker。

用起来怎样，种子都下得动吗

hteane, 2021-03-04:
基本都下的动，我在rarbg找的磁力，总体来说就是开了3天电信给我发短信说超了1T限流给我降速到500M带宽了。。。

hteane, 2021-03-04:
今天0点开始到现在下载80g上传20g

Neilson8, 2021-03-04:
> 今天0点开始到现在下载80g上传20g

你QQ多少，我准备安装 了，有问题问下你

Neilson8, 2021-03-04:
> > 有源码链接吗？这个必须搞，方便把源码留在这，明天我开搞，后天才能用
> 
> 有办法增加小米路由器那个迅雷下载插件吗，那个很好用，跟迅雷差不多，支持很多格式，种子都下得动

https://github.com/qbittorrent/qBittorrent/releases 源码找到了

Neilson8, 2021-03-05:
> 先运行命令“docker pull johngong/qbittorrent:qee-latest”下载镜像，这个是集成额外tracker的版本。然后去容器里新建容器-输入命令：
> docker create
> --name=qbittorrent
> -e WEBUIPORT=8989
> -e UMASK=022 \
> -e UID=65534 [...]

在吗，我安装 好了，启动了，也能进去，但是没有地址选网络HOST的，面板那提示缺少BT TRACKER?

hteane, 2021-03-05:
输入完命令后的配置页面下拉会有网络选项，默认是bridge，改成host就行，不然连不上网。

hteane, 2021-03-05:
你要先停止容器然后移除重新添加容器，容器运行后好像不能修改。在运行状态里好像可以断开目前的bridge连接然后新增host连接，但是我没用过，保险一点重新新建容器吧。

Neilson8, 2021-03-05:
> 停止容器然后移除重新添加容器，容器运行后好像不能修改。在运行状态里好像可以断开目前的bridge连接然后新增host连接，但是我没用过，保险一点重新新建容器吧。

你QQ多少给我啊

Neilson8, 2021-03-05:
> > 停止容器然后移除重新添加容器，容器运行后好像不能修改。在运行状态里好像可以断开目前的bridge连接然后新增host连接，但是我没用过，保险一点重新新建容器吧。
> 
> 你QQ多少给我啊

> 你要先停止容器然后移除重新添加容器，容器运行后好像不能修改。在运行状态里好像可以断开目前的bridge连接然后新增host连接，但是我没用过，保险一点重新新建容器吧。

有速度，但是只下载一会就停了，很慢，怎么破

closed 2021-03-19 by DHDAXCW