DHDAXCW/NanoPi-R4S-R4SE #35: 今天这15版固件网口突然互换了，已删除，用执行代码取消互换，改回原来的

DHDAXCW, 2021-03-15:


DHDAXCW, 2021-03-15:
原因是lean源码的网口没有被执行，这样不生效，已用代码执行把互换取消了，改回原来的。这样r4s用互换完全没有意义、r2s可以互换。

bg4drl, 2021-03-15:
侧面追加一句, R4S不互换的话 每启动一次 Mac地址就在跳跃  我现在也用互换.

DHDAXCW, 2021-03-15:
> 侧面追加一句, R4S不互换的话 每启动一次 Mac地址就在跳跃 我现在也用互换.

我先把docker内核解决再处理你问题

ljtao020, 2021-03-15:
> 侧面追加一句, R4S不互换的话 每启动一次 Mac地址就在跳跃 我现在也用互换.

确实每次重启路由，电脑都会发现新网络，原来是这个原因，互换就可以解决了吗？

lizhi0506, 2021-03-18:
用你固件里边的open class，为什么网络测速总是很低？换用Pass wall测速就正常了

lizhi0506, 2021-03-18:
跳到内容
无需任何代码即可学习Git和GitHub！
使用“ Hello World”指南，您将开始分支，编写注释并打开拉取请求。

DHDAXCW
/
NanoPi-R4S-2021
代码
问题
31
拉取要求
动作
专案
维基
安全
见解
今天这15版固件网口突然互换了，已删除，用执行代码取消互换，改回原来的 ＃35
 打开
DHDAXCW 打开了此问题 3天前 ·5条评论
 打开
今天这15版固件网口突然互换了，已删除，用执行代码取消互换，改回原来的
＃35
DHDAXCW 打开了此问题 3天前 ·5条评论
评论
 
所有者
DHDAXCW 已评论 3天前
没有提供描述。

@DHDAXCW DHDAXCW 已评论 3天前
原因是lean令牌的网口没有执行，这样不执行，已用代码执行把互换取消了，改回原来的。这样r4s用互换完全没有意义，r2s可以互换。

@ bg4drl bg4drl 已评论 3天前
侧面追加一句，R4S不互换的话每启动一次Mac地址就在跳跃我现在也用互换。

@DHDAXCW DHDAXCW 已评论 3天前
侧面追加一句，R4S不互换的话每启动一次Mac地址就在跳跃我现在也用互换。

我先把docker内核解决再处理你问题

@ ljtao020 ljtao020 已评论 3天前
侧面追加一句，R4S不互换的话每启动一次Mac地址就在跳跃我现在也用互换。

确实重新启动路由，电脑都会发现新网络，原来是这个原因，互换就可以解决了吗？

 
lizhi0506 已评论 11秒钟前
用你固件里边的公开课，为什么网络测速总是很低？换用Pass wall测速就正常了

发表评论
请记住，对此存储库的贡献应遵循我们的 GitHub社区准则。
受让人
没有人分配
标签
还没有
专案
还没有
里程碑
没有里程碑
链接的拉取请求
成功合并拉取请求可能会解决此问题。

还没有
通知事项
定制
您收到评论是因为您发表了评论。
4名参加者
@ bg4drl
@ ljtao020
@ lizhi0506
@DHDAXCW
©2021 GitHub，Inc.
条款
隐私
安全
状态
文件
联系GitHub
价钱
原料药
训练
博客
关于
lizhi0506评论了现在用你固件里边的openclash，为什么网络测速总是很低？换用Pass wall测速就正常了

closed 2021-03-19 by DHDAXCW