DHDAXCW/NanoPi-R4S-R4SE #39: Transmission开启后，R4S重启后不能正常启动

Game4587, 2021-03-19:
刷完固件后，Transmission设置完毕，如果没下载任务东西，R4S重启，Transmission可以正常启动。
如果Transmission下载过东西，R4S重启后,Transmission就不能启动了，在Transmission设置界面没有打开控制面版的选项，输入192.168.2.1:9091后打开Transmission的web界面，只能开启到一半，里面只显示工具条，左边的菜单是空的，有时会出拒绝访问。
在TTYD使用 /etc/init.d/transmission stop 这条命令可以使用，使用安装的命令却opkg_install_cmd: Cannot install package transmision-daemo-openssl
难道每次都要重刷固件吗,还不能重启R4S????

Game4587, 2021-03-19:
另外我发现如果Transmission正下载时重启R4S，Transmission会启动不了，而且R4S的CPU使用率会一直到100%，重启也没法停

DHDAXCW, 2021-03-19:
软重启？

DHDAXCW, 2021-03-19:
谢谢反馈啊
软重启情况下我是不能保证它是否正常运作。
我今天又再优化重启问题，明天可以尝试。
另外，TF卡也要好的，还是闪迪容易有问题，我目前用的是三星
刚才我帮你试了，下载和没下载情况下，顺便看了重启后CPU没有满载，没有你说那个问题。
要不这样吧，直接拔电可以试试
最后提供一下版本日期。

Game4587, 2021-03-19:
使用的是03.13和12的这两个，TF卡0313版使用的是闪迪A2 128的，0312版用的是三星12G，然后USB插了个西数固态1T硬盘，
挂载点是/mnt/sda1，里面设置了Download、Media、Settings目录，都设了777权限,Transmission的配置文件的目录在Settings,下载目录在Download,
最新版本好想刷，但是现在这个时候，如果我敢刷机的话，R4S会被老婆丢楼下去的，只能明天再试试新版本了··

DHDAXCW, 2021-03-19:
哈哈这样啊，你这设置有点过分，这电压能带的动吗？移动硬盘需要单独供电，不然让r4s给他供电，这样波动会不稳定。正好我给你看看我在另一个群里看到这玩意有点大（图在下面）
明天20版本不保证是否正常，你最好下午刷，因为白天我要测试固件是否稳定。如果我没有删除固件说明正常。[image]

Game4587, 2021-03-23:
找到原因了，如果只是全局设置，启用Transmission，设置下载目录和设置目录，这三个，重启后还可以一直正常使用。
如果除全局使用外，后面的几页设置也设置了或改变了参数，像RPG开了验证，那么R4S重启后，就不能正常进入192.168.2.1:9091页面，transmission不能正常启动。系统日志里面显示
Sun Mar 21 18:58:48 2021 daemon.err transmission-daemon[7679]: [2021-03-21 18:58:48.189] JSON parse failed in /mnt/sda1/Settings//settings.json at pos 877: INVALID_NUMBER -- remaining text "KB/s,
Sun Mar 21 18:58:48 2021 daemon.err transmission-daemon[7679]: "speed-lim"
Sun Mar 21 18:58:48 2021 daemon.err transmission-daemon[7679]: [2021-03-21 18:58:48.190] transmission-daemon Error loading config file -- exiting. (daemon.c:693)