DHDAXCW/NanoPi-R4S-R4SE #47: 2021.03.13 Lean OpenWrt以后的版本RTL8812BU网卡驱动不了

aimuch, 2021-03-31:
如题，3.13版本RTL8812BU网卡完美驱动，但以后的版本WiFi用不了了；
另外，wan口的Mac地址更改一直无效，vim /etc/config/network更改wan口的地址才有效

DHDAXCW, 2021-03-31:
> 如题，3.13版本RTL8812BU网卡完美驱动，但以后的版本WiFi用不了了；
> 另外，wan口的Mac地址更改一直无效，vim /etc/config/network更改wan口的地址才有效

这个版本开始，我一直在解决编译问题，等编译成功了 我就开始把驱动打上最新版。之前版本是驱动源失效了。现在尝试能不能编译成功。如果编译成功。我就打上

aimuch, 2021-03-31:
> > 如题，3.13版本RTL8812BU网卡完美驱动，但以后的版本WiFi用不了了；
> > 另外，wan口的Mac地址更改一直无效，vim /etc/config/network更改wan口的地址才有效
> 
> 这个版本开始，我一直在解决编译问题，等编译成功了 我就开始把驱动打上最新版。之前版本是驱动源失效了。现在尝试能不能编译成功。如果编译成功。我就打上

辛苦辛苦

aimuch, 2021-04-01:
> > 如题，3.13版本RTL8812BU网卡完美驱动，但以后的版本WiFi用不了了；
> > 另外，wan口的Mac地址更改一直无效，vim /etc/config/network更改wan口的地址才有效
> 
> 这个版本开始，我一直在解决编译问题，等编译成功了 我就开始把驱动打上最新版。之前版本是驱动源失效了。现在尝试能不能编译成功。如果编译成功。我就打上

4.1 版本RTL8812BU网卡完美驱动了，*wan口的Mac地址更改一直无效，vim /etc/config/network更改wan口的地址才有效*有待解决

DHDAXCW, 2021-04-01:
> > > 如题，3.13版本RTL8812BU网卡完美驱动，但以后的版本WiFi用不了了；
> > > 另外，wan口的Mac地址更改一直无效，vim /etc/config/network更改wan口的地址才有效
> > 
> > 这个版本开始，我一直在解决编译问题，等编译成功了 我就开始把驱动打上最新版。之前版本是驱动源失效了。现在尝试能不能编译成功。如果编译成功。我就打上
> 
> 4.1 版本RTL8812BU网卡完美驱动了，*wan口的Mac地址更改一直无效，vim /etc/config/network更改wan口的地址才有效*有待解决

这个我今晚检查，好不容易解决编译失败问题...

aimuch, 2021-04-01:
> > > > 如题，3.13版本RTL8812BU网卡完美驱动，但以后的版本WiFi用不了了；
> > > > 另外，wan口的Mac地址更改一直无效，vim /etc/config/network更改wan口的地址才有效
> > > 
> > > 
> > > 这个版本开始，我一直在解决编译问题，等编译成功了 我就开始把驱动打上最新版。之前版本是驱动源失效了。现在尝试能不能编译成功。如果编译成功。我就打上
> > 
> > 
> > 4.1 版本RTL8812BU网卡完美驱动了，_wan口的Mac地址更改一直无效，vim /etc/config/network更改wan口的地址才有效_有待解决
> 
> 这个我今晚检查，好不容易解决编译失败问题...

辛苦了，我在网页端修改wan口的mac地址其实是生效了，vim /etc/config/network里看wan_eth0_dev没生效，直接删除然后重启就可以了。
```
onfig interface 'wan'
        option ifname 'eth0'
        option proto 'dhcp'
        option macaddr 'a0:8c:11:de:5b:11'

config device 'wan_eth0_dev'
        option name 'eth0'
        option macaddr 'fe:5b:6e:63:06:78'

config interface 'wan6'
        option ifname 'eth0'
        option proto 'dhcpv6'

config interface 'vpn0'
        option ifname 'tun0'
        option proto 'none'
```

DHDAXCW, 2021-04-01:
是不是把wan_eth0_dev这个删除，然后重启，就可以直接在web上改了？

aimuch, 2021-04-01:
> 是不是把wan_eth0_dev这个删除，然后重启，就可以直接在web上改了？

对的，刚刚试过，确实是这样的

DHDAXCW, 2021-04-01:
> > 是不是把wan_eth0_dev这个删除，然后重启，就可以直接在web上改了？
> 
> 对的，刚刚试过，确实是这样的

好的 谢谢提供啊 我今晚开搞