DHDAXCW/NanoPi-R4S-R4SE #53: cloudfare cdn解析 部分网站不正常

dddfgeer, 2021-04-06:
打开有关cloudfare cdn托管的网站会显示error1020 或者部分显示禁止中国ip访问 这些问题是在电脑浏览器才有这些问题 手机连接wifi没有这些问题  麻烦解决一下 谢谢

dddfgeer, 2021-04-06:
duyaoss.com 这些网站就是cloudfare cdn的网站 还有其他网站cloudfare cdn托管的都有这些问题

dddfgeer, 2021-04-06:
@

- [ ] 

1. 

- 

> ### ### **_`[](url)`_**

dddfgeer, 2021-04-07:
这些问题在passwall ssrplus 还有hello world都有这些问题

dddfgeer, 2021-04-07:
而且在电脑浏览器打开cloudfare cdn托管的网站被识别成中国大陆ip https://www.cloudflare.com/cdn-cgi/trace 手机正常

dddfgeer, 2021-04-07:
还有个毛病就是部分国外冷门网站无法解析

dddfgeer, 2021-04-07:
> 打开有关cloudfare cdn托管的网站会显示error1020 或者部分显示禁止中国ip访问 这些问题是在电脑浏览器才有这些问题 手机连接wifi没有这些问题 麻烦解决一下 谢谢

dddfgeer, 2021-04-07:
> 打开有关cloudfare cdn托管的网站会显示error1020 或者部分显示禁止中国ip访问 这些问题是在电脑浏览器才有这些问题 手机连接wifi没有这些问题 打开有关cloudfare cdn托管的网站会显示error1020 或者部分显示禁止中国ip访问 这些问题是在电脑浏览器才有这些问题 手机连接wifi没有这些问题 duyaoss.com 这些网站就是cloudfare cdn的网站 还有其他网站cloudfare cdn托管的都有这些问题
passwall ssrplus 还有hello world都是这些情况 而且在电脑浏览器打开cloudfare cdn托管的网站被识别成中国大陆ip https://www.cloudflare.com/cdn-cgi/trace 手机正常 

@dddfgeer
 
Author
dddfgeer commented 9 minutes ago
打开有关cloudfare cdn托管的网站会显示error1020 或者部分显示禁止中国ip访问 这些问题是在电脑浏览器才有这些问题 手机连接wifi没有这些问题 麻烦解决一下 谢谢

DHDAXCW, 2021-04-07:
不好意思，我母鸡

dddfgeer, 2021-04-07:
好的 发截图给你

dddfgeer, 2021-04-07:
[image]

dddfgeer, 2021-04-07:
[image]

dddfgeer, 2021-04-07:
这些毛病是在windows电脑浏览器才有

dddfgeer, 2021-04-07:
就是这些网站可能是dns解析的问题 cloudfare cdn托管的 手机浏览器打开正常 在电脑浏览器打开显示中国大陆ip

geitwhat, 2021-05-14:
> 托管的手机浏览

我也发现这个问题 有办法解决吗

geitwhat, 2021-05-14:
> 就是这些网站可能是dns解析的问题cloudfare cdn托管的手机浏览器打开正常在电脑浏览器打开显示中国大陆ip
我也发现这个问题有办法解决吗