DHDAXCW/NanoPi-R4S-R4SE #59: R4s加装什么规格的风扇？

S1U, 2021-04-13:
有没有推荐？谢谢。

DHDAXCW, 2021-04-13:
> 有没有推荐？谢谢。

这...超频超热了？要不，把超频移除了吧？

S1U, 2021-04-14:
最近几个固件刷了，aria2老是出错下载，老是提示“发生 I/O 错误”。要重启路由，才会继续，过了段时间又出错。怀疑是太热。

ljtao020, 2021-04-17:
> 最近几个固件刷了，aria2老是出错下载，老是提示“发生 I/O 错误”。要重启路由，才会继续，过了段时间又出错。
> 刚看了下(2208MHz 1416MHz, 42.2°C)。
> 
> 这原因是不是太热？

才42嘛，肯定不是过热了

DHDAXCW, 2021-04-17:
> 最近几个固件刷了，aria2老是出错下载，老是提示“发生 I/O 错误”。要重启路由，才会继续，过了段时间又出错。
> 刚看了下(2208MHz 1416MHz, 42.2°C)。
> 
> 这原因是不是太热？

你咋把CPU拉满了....你电源跑的功耗多少？

xurenzhong1995, 2021-04-18:
42度就叫热？温度低于60度都是正常的

icm2203, 2021-04-19:
> 最近几个固件刷了，aria2老是错误下载，老是提示“发生I / O错误”。要重启路由，才会继续，过了段时间又出错。
> 刚看了下（2208MHz 1416MHz，42.2°C ）。
> 
> 这原因是不是太热？

你还是换卡吧，真不是超频的事儿。看我发的不超频的温度。

S1U, 2021-04-20:
> > 最近几个固件刷了，aria2老是出错下载，老是提示“发生 I/O 错误”。要重启路由，才会继续，过了段时间又出错。
> > 刚看了下(2208MHz 1416MHz, 42.2°C)。
> > 这原因是不是太热？
> 
> 你咋把CPU拉满了....你电源跑的功耗多少？

没动固件里cpu设置，好像是更新了某一天的固件之后发生的。现在更新了`DHDAXCW build 2021.04.18 @ FusionWrt R21.3.27 (2021.04.18) / LuCI Master (git-21.096.35788-b200c7b)`这个版本到现在还没出现问题。。。