DHDAXCW/NanoPi-R4S-R4SE #69: passwall开启负载均衡没作用

icm2203, 2021-04-24:
开启负载均衡
[image]

主节点里确没有开启
[image]

wangkai6688, 2021-04-24:
@icm2203
#### 这个不是固件问题，负载均衡打开后要自己设置节点

- 添加一个节点，类型和你负载均衡中添加的**节点**保持一致，地址127.0.0.1 ，密码和**节点**的密码保持一致(注意**不是**负载均衡的密码)，端口填**负载均衡端口**(默认1181)，其余配置和**节点**保持一致 如下图
[image]

icm2203, 2021-04-27:
> @ icm2203
> 
> #### 这个不是固件问题，负载均衡打开后要自己设置例程
> * 添加一个节点，状语从句：类型你负载均衡中添加的**节点**保持harmony和谐，地址127.0.0.1，状语从句：密码**节点**的密码保持harmony和谐（注意**不是**负载均衡的密码），填端口**负载均衡端口**（默认1181），其余配置和**议员**保持一致如下图
>   [image]

新版需要手动设置了啊！好的知道了，谢谢。