DHDAXCW/NanoPi-R4S-R4SE #81: 端口转发无效，精简版不能通过sftp和scp远程连接

themanwhohehe, 2021-05-21:
刷了标准版，非超频版和精简版，防火墙的端口转发无法正常工作，我主要是通过互联网用moonlight串流访问家里电脑，每次都是勉强接上，然后两秒后就退出。并且精简版无法使用transmit通过sftp和scp远程连接，后来刷了DHDAXCW二月份的friendly官方固件才解决了这个问题，好奇怪，特此报告一下

cjwddtc, 2021-05-21:
不建议直接暴露服务到公网上，尤其的moonlight这种可以直接控制电脑的服务，最好是通过vpn转一下，vpn协议毕竟设计的时候会十分充分的考虑安全性（至少比nvidia的串流协议考虑的更充分）。

cjwddtc, 2021-05-21:
感觉moonlight很卡，我连回家的延迟10ms也感觉卡，就很难受。

themanwhohehe, 2021-05-21:
> 不建议直接暴露服务到公网上，尤其的moonlight这种可以直接控制电脑的服务，最好是通过vpn转一下，vpn协议毕竟设计的时候会十分充分的考虑安全性（至少比nvidia的串流协议考虑的更充分）。

可以用Zerotier，但我使用端口转发，习惯了

wangkai6688, 2021-06-06:
我发现好像是SFE Fast Path 加速的问题

hardyhuang818, 2021-06-06:
请问 DHDAXCW build 2021.04.19 @ FusionWrt R21.4.18 (2021.04.19) / LuCI Master
(git-21.096.35788-b200c7b)   这个版本端口转发 可以用吗？

wangkai6688, 2021-06-06:
> […](#)

我用着是可以的，现在也不是不行，就是好像有一些协议不能完全通过

hardyhuang818, 2021-06-06:
感谢感谢！刚把“ 只允许内网访问  ”勾取消掉了就可以转发成功了...