DHDAXCW/NanoPi-R4S-R4SE #84: 联通500兆宽带通过openwrt系统变成1百兆不到，请教大佬原因

quango-web, 2021-06-16:
硬件构成：R4S主（大佬614固件，基本默认设置）+网件R7800（AP模式）
原使用电信一百兆宽带，测速可达到150～160以上。
1、今天换上联通5百兆（华为光猫WO_27s：联通默认设置为路由WAN模式），软路由WAN口设为DHCP客户端模式，测速不到百兆。
2、进入光猫修改为桥接WAN模式，软路由设为pppoe拨号模式，测速依然百兆不足。
3、将光猫（路由模式）直连网件X4S（AP模式），测速可达450～550兆。

在此请教大佬们如何解决此疑难杂症，谢谢！

oniLand, 2021-06-18:
我之前也遇到过这个问题，使用dmesmg查看内核日志
[image]
eth0 最后掉速到了100mbps全双工
使用如图的命令查看一下对应接口的速率
[image]

如果掉速
使用
ethtool -s eth0 speed 1000 duplex full autoneg off
手动关闭自动协商

oniLand, 2021-06-18:
不知道是否是网卡驱动问题。

oniLand, 2021-06-18:
版本 DHDAXCW's FusionWrt built on 2021.06.10 docker

quango-web, 2021-06-19:
TTYD终端使用命令关闭自动协商后显示speed:千兆/s，但直接断网了。重启路由后又自动恢复speed:百兆/s
[image]

quango-web, 2021-06-19:
改用电信宽带光猫（百兆套餐，速率达到160MBps）同一个网线连接R4S拨号，TTYD直接显示speed:千兆/s
[image]

quango-web, 2021-06-19:
> 不知道是否是网卡驱动问题。

用电信跟联通光猫各试了几次，又好似不像openwrt固件网卡驱动问题，头大

paradox73, 2021-09-04:
请问问题最后是如何解决的？我也是R4S+联通500  没加R4S之前可以跑满，加了之后限制在1百兆内了，用的是星最多那版固件

oniLand, 2021-09-04:
我这边后面判断是可能光猫到r4s 的wan口网线太垃圾了，买的杂牌的，后来我全部换上了正规牌子的7类网线就在没出现过这种情况了。

bluebabes, 2021-11-24:
还是关了吧， 我也出现了。

一直以为是光猫问题，速度慢了，重启光猫就正常了， 今天偶尔看到openwrt后台，竟然是 100M 双工

可能桥接的自动协商有点问题

还是按 oniLand 说的手动关闭自动协商靠谱

ethtool -s eth0 speed 1000 duplex full autoneg off

remychew, 2021-12-12:
我也同样出现了这个问题，经常掉到100M，插上USB这个问题出现的更频繁。关闭自动协商之后，好像就断网了 @bluebabes 你的没有问题吗？

Wei-Xia, 2022-01-08:
@quango-web  你好，我也有同样的问题，请问后来排除出来的问题所在吗？

remychew, 2022-01-08:
@Wei-Xia 我解决了，桥接模式下，联通的光猫跟R4S的网口速率握手可能存在问题，将桥接模式关闭，改成使用光猫拨号，停止使用R4S的USB设备，我没有再遇到速率下降的问题。

Wei-Xia, 2022-01-08:
@remychew   谢谢！我本身就是光猫拨号，eth0 的上限一直是 100Mb/s。后来，我把软路由接在光猫上的 LAN 口换了另一个口，问题就解决了。。。。

ymy0819, 2022-02-07:
我的也是这个问题，我电信500m刷了好多个版本的固件都不行，速度只有100m，比如骷髅固件自己掉速，小朋友固件的老版没有问题速度正常，但最新版反到速度只有100m，不知道各位最后都是怎么解决的？关闭全双功的方法好用不？

zgc, 2022-02-07:
我也遇到这个问题，感觉是旧款光猫的原因，我换个新光猫就没遇到了

Wei-Xia, 2022-02-07:
我换了网线和接口就好了，不是软件的问题，是硬件的问题。

chandlerniu, 2022-03-27:
我也遇到这样问题了，一直以为是固件问题，刷了之后还是出现这样错误，忽然想到光猫没有桥接前没有这种问题，光猫和软路由之间有一段距离，用的是之前装修预埋的网线，原因就出在这里，网线质量一般，将软路由直接放到光猫箱里后这个问题就自动消失了！

chandlerniu, 2022-03-29:
补充一下，网线过长和质量一般都会出现这种问题！放到光猫配电箱里网线长度很短，并且质量好，问题解决！

Ymy0825, 2022-06-06:
> 还是关了吧， 我也出现了。
> 
> 一直以为是光猫问题，速度慢了，重启光猫就正常了， 今天偶尔看到openwrt后台，竟然是 100M 双工
> 
> 可能桥接的自动协商有点问题
> 
> 还是按 oniLand 说的手动关闭自动协商靠谱
> 
> ethtool -s eth0 speed 1000 duplex full autoneg off

你好，如何手动关闭自动协商呀？我用ethtool -s eth0 speed 1000 duplex full autoneg off，这个命令后直接不能上网了

chandlerniu, 2022-06-06:
我尝试关掉后依然没有解决，还是搞定网线才能根本解决问题

Ymy0825, 2022-06-06:
好的，我换网线试试效果

xeasonwang, 2022-07-03:
> 好的，我换网线试试效果。看来这R4S挺挑网线呀，我原来一直用的华硕路由器就不会有这样的问题。

flow control，关掉 shortcut-FE 试试

hww-png, 2022-10-26:
请问你的问题解决了吗？我现在问题和你描述的差不多，我是软路由拨号，和软路由桥接的硬路由协商只有100M。但是换成硬路由拨号软路由做旁路网关就没有这个问题

remychew, 2022-10-27:
你可以尝试更换网线试一下，如果没有解决，那就是软路由和光猫的协商有问题。
我现在用了笨方法，不过也是有效的，就是由硬路由桥接拨号，软路由作为二级路由。

remychew, 2022-11-25:
各位，一直没有解决桥接模式掉速问题的，排除网线质量后，可能是跟R4S温度过高有关，你们是否有添加风扇给R4S降温？
我加上风扇后，使用桥接模式，再也没有出现掉速问题。

jaosnsmy, 2023-03-15:
我也遇到了同样的问题，自己研究了一晚上解决了。我是r4s 联通700兆宽带，光猫拨号，软路由开启dhcp，硬路由做桥接，USB挂在移动硬盘做扩展。
解决方法如下：尝试对调WAN口和LAN口的接口，WAN更改为eth1，LAN更改为eth0，更改后WAN口变为1000兆双工。大家可以尝试一下这个方法

jaosnsmy, 2023-03-15:
> 我也遇到了同样的问题，自己研究了一晚上解决了。我是r4s 联通700兆宽带，光猫拨号，软路由开启dhcp，硬路由做桥接，USB挂在移动硬盘做扩展。 解决方法如下：尝试对调WAN口和LAN口的接口，WAN更改为eth1，LAN更改为eth0，更改后WAN口变为1000兆双工。大家可以尝试一下这个方法

经测试变更网口后，挂载USB硬盘后再次出现速率变为100兆的情况，再次调换网口后又恢复1000兆，初步判断为挂在USB硬盘对速率有影响。

jaosnsmy, 2023-03-16:
我也遇到了同样的问题，自己研究了一晚上解决了。我是r4s 联通700兆宽带，光猫拨号，软路由开启dhcp，硬路由做桥接，USB挂在移动硬盘做扩展。

jaosnsmy, 2023-03-16:
> 我也遇到了同样的问题，自己研究了一晚上解决了。我是r4s 联通700兆宽带，光猫拨号，软路由开启dhcp，硬路由做桥接，USB挂在移动硬盘做扩展。

后来把墙上连接软路由和光猫之间的网线换了一根就好了，原来连接的是一根七类网线，墙中预埋的是超五类网线，后来我连接的线换成普通的超五类网线就自动变成千兆全双工了，如果有遇到同样问题的伙伴可以试一下更换连接光猫和软路由wan口的那根网线，大概率是那根线的问题

MAOGE2018, 2023-05-13:
晚点试试，这问题真头疼，有时候重启可以有时候不行。

jaosnsmy, 2023-08-11:
是后者的 入户连接光猫 然后进墙预埋长距离网线 网线在墙面预留了网线接口面板  我指的换线是从墙面预留的网线接口面板连接到软路由的线 

---- 回复的原邮件 ----
| 发件人 | ***@***.***> |
| 发送日期 | 2023年08月11日 20:54 |
| 收件人 | DHDAXCW/NanoPi-R4S-R4SE ***@***.***> |
| 抄送人 | jaosnsmy ***@***.***>,
Comment ***@***.***> |
| 主题 | Re: [DHDAXCW/NanoPi-R4S-R4SE] 联通500兆宽带通过openwrt系统变成1百兆不到，请教大佬原因 (#84) |

我也遇到了同样的问题，自己研究了一晚上解决了。我是r4s 联通700兆宽带，光猫拨号，软路由开启dhcp，硬路由做桥接，USB挂在移动硬盘做扩展。

后来把墙上连接软路由和光猫之间的网线换了一根就好了，原来连接的是一根七类网线，墙中预埋的是超五类网线，后来我连接的线换成普通的超五类网线就自动变成千兆全双工了，如果有遇到同样问题的伙伴可以试一下更换连接光猫和软路由wan口的那根网线，大概率是那根线的问题

您把我说迷糊了，您的光猫是直接【近距离】连接的软路由，还是【先走的墙内预埋的网线之后】再走你的网线连的软路由啊？

应该是后者？

saveurass, 2023-08-12:
> 我也遇到这个问题，感觉是旧款光猫的原因，我换个新光猫就没遇到了

这个是唯一的正解，换网线没用，只有换光猫才有用，新出来的光猫解决了问题