DHDAXCW/NanoPi-R4S-R4SE #87: 凌晨4点自动关机

45142918, 2021-07-03:
固件是2021.06.29 r4s fdrmal edition_正式版_高大全
连续2晚都是凌晨4点自动关机
早上刷了2021.07.03 r4s fdrmal edition_正式版_高大全
再观察下还会不会有这个情况

mouse0907, 2021-07-06:
我的也是，每天早上自动关机了~请问在那里设置的啊？可以勾选掉么？

45142918, 2021-07-06:
> 我的也是，今天早上自动恢复了~请问在那里的啊？可以勾选掉么？

我没最终解决，刷了其他固件。

mouse0907, 2021-07-06:
我刷了稳定版最新固件，今天测试后还是一晚自动关机了。请问你后面用了什么固件解决了此问题的再次发生。

45142918, 2021-07-07:
> 我刷了稳定版最新固件，今天测试后还是一晚自动关机了。请问你后面用了什么固件解决了此问题的再次发生。

https://www.right.com.cn/forum/thread-4083701-1-1.html 刷了这个

mouse0907, 2021-07-07:
好的，谢谢，我试试看~

mouse0907, 2021-07-09:
这个固件很简单，里面很多插件源没有~需要手动添加。凌晨关机目前还没有测试。今天晚上测试后反馈情况。

DHDAXCW, 2021-07-09:
7.4以上固件改了好多配置，这不会存在问题了

mouse0907, 2021-07-10:
> 7.4以上固件改了好多配置，这不会存在问题了

多谢骷髅头大大~，我今天晚上用新TF卡再刷这个版本试试看。我测试了一下，0328版本的固件昨天4点没有重启，一切正常。

45142918, 2021-07-10:
> > 7.4以上固件改了好多配置，这不会存在问题了
> 
> 多谢骷髅头大大~，我今天晚上用新TF卡再刷0708这个版本试试看。我测试了一下，0328版本的固件昨天4点没有重启，一切正常。

测试后反馈下，如果正常了，我也刷回

mouse0907, 2021-07-10:
> > > 7.4 以上的改造了很多配置，这不会存在的问题了
> > 
> > 
> > 多谢骷髅头大~，今天晚上用新TF卡再刷0708这个版本我看看。我测试一下，0328版本的周末昨天4点没有重启，一切正常。
> 
> 测试后反馈下，如果正常了，我也刷回

好的，没问题。我看到现在只是精简版的，我先刷好测试。结果明天告诉你。以后有高大全再看看情况。

mouse0907, 2021-07-10:
> > > 7.4以上固件改了好多配置，这不会存在问题了
> > 
> > 
> > 多谢骷髅头大大~，我今天晚上用新TF卡再刷0708这个版本试试看。我测试了一下，0328版本的固件昨天4点没有重启，一切正常。
> 
> 测试后反馈下，如果正常了，我也刷回

测试失败0708固件4点关机

mouse0907, 2021-07-10:
> 7.4以上固件改了好多配置，这不会存在问题了

骷髅头大大，0708稳定精简版已经测试，故障依旧。凌晨4点关机。用了三星和海康的卡。目前怀疑是diskman的关系。我在测试一下。

45142918, 2021-07-12:
> > 7.4以上固件改了好多配置，这不会存在问题了
> 
> 骷髅头大大，0708稳定精简版已经测试，故障依旧。凌晨4点关机。用了三星和海康的卡。目前怀疑是diskman的关系。我在测试一下。

测试结果怎样？

mouse0907, 2021-07-12:
> > > 7.4以上固件改了好多配置，这不会存在问题了
> > 
> > 
> > 骷髅头大大，0708稳定精简版已经测试，故障依旧。凌晨4点关机。用了三星和海康的卡。目前怀疑是diskman的关系。我在测试一下。
> 
> 测试结果怎样？

故障依旧，继续等新版本

DHDAXCW, 2021-07-12:
这个版本我跑了一天cpu测试。等没有出现死机。等凌晨我再做测试一下吧

45142918, 2021-07-12:
> 这个版本我跑了一天cpu测试。没有出现死机。等凌晨我再做测试一下吧

白天是不会有问题的，就是凌晨4点会自动关机

mouse0907, 2021-07-12:
> 这个版本我跑了一天cpu测试。没有出现死机。等凌晨我再做测试一下吧

骷髅头大大，我感觉不是死机问题，很有可能是某个插件的原始设置问题，或者是那个插件BUG了。因为每天都是4点关机，很准时。我现在准备在白天把R4S的时间调整到3:58,然后看情况

mouse0907, 2021-07-12:
并且，是从0406这个版本开始的，我测试过了

mouse0907, 2021-07-12:
上班摸鱼，用修改时间打法测试了一下，每天凌晨2点有一批插件有动作，随后凌晨4点有“一大批“插件有动作，并在4:00:13自动重启设备，测试的时候设备重启成功。估摸着如果重启不成功，就自动关机了。由于重启以后就看不到日志内容了，故抓不到是那个插件，希望骷髅头大大找到，并关闭自动重启，次bug应该就会攻破了。是不是在0328到0405版本之间更新了哪些插件呀？估计问题就在更新的插件中。该测试仅供参考，只能做这么微不足道的事情了。

DHDAXCW, 2021-07-12:
> 上班摸鱼，用修改时间打法测试了一下，每天凌晨2点有一批插件有动作，随后凌晨4点有“一大批“插件有动作，并在4:00:13自动重启设备，测试的时候设备重启成功。估摸着如果重启不成功，就自动关机了。由于重启以后就看不到日志内容了，故抓不到是那个插件，希望骷髅头大大找到，并关闭自动重启，次bug应该就会攻破了。是不是在0328到0405版本之间更新了哪些插件呀？估计问题就在更新的插件中。该测试仅供参考，只能做这么微不足道的事情了。

凌晨4点跑CPU测试的。这个我也发现了，可是我还没死机...我尽量在环境条件下测试试试吧

45142918, 2021-07-12:
> > 上班摸鱼，用修改时间打法测试了一下，每天凌晨2点有一批插件有动作，随后凌晨4点有“一大批“插件有动作，并在4:00:13自动重启设备，测试的时候设备重启成功。估摸着如果重启不成功，就自动关机了。由于重启以后就看不到日志内容了，故抓不到是那个插件，希望骷髅头大大找到，并关闭自动重启，次bug应该就会攻破了。是不是在0328到0405版本之间更新了哪些插件呀？估计问题就在更新的插件中。该测试仅供参考，只能做这么微不足道的事情了。
> 
> 凌晨4点跑CPU测试的。这个我也发现了，可是我还没死机...我尽量在环境条件下测试试试吧

期待修复

45142918, 2021-07-12:
> 上班摸鱼，用修改时间大法测试了一下R4S模拟每个整点的动作。现在情况是这样的。
> 1、每天凌晨2点有一批插件有动作，3点无事。
> 2、随后凌晨4点有“一大批“插件有动作，并在4:00:13自动重启设备，本次测试的时候设备重启成功。
> 3、估摸着如果重启不成功，就自动关机了。
> 由于重启以后就看不到日志内容了，故抓不到是那个插件，希望骷髅头大大找到，并关闭自动重启，次bug应该就会攻破了。是不是在0328到0405版本之间更新了哪些插件呀？估计问题就在更新的插件中。
> 
> 该测试仅供参考，只能做这么微不足道的事情了。

45142918, 2021-07-14:
> > 上班摸鱼，用修改时间大法测试了一下R4S模拟每个整点的动作。现在情况是这样的。
> > 1、每天凌晨2点有一批插件有动作，3点无事。
> > 2、随后凌晨4点有“一大批“插件有动作，并在4:00:13自动重启设备，本次测试的时候设备重启成功。
> > 3、估摸着如果重启不成功，就自动关机了。
> > 由于重启以后就看不到日志内容了，故抓不到是那个插件，希望骷髅头大大找到，并关闭自动重启，次bug应该就会攻破了。是不是在0328到0405版本之间更新了哪些插件呀？估计问题就在更新的插件中。
> > 该测试仅供参考，只能做这么微不足道的事情了。

有没有测试到哪个版本的可以？

DHDAXCW, 2021-07-14:
已修复，结束

45142918, 2021-07-14:
> 已修复，结束
> 明天版本发布。

收到，期待，感谢大大

mouse0907, 2021-07-16:
测试了一次最新0714无超频版本，还是4点关机📴，问题没解决。今晚继续测试。

NathanDai, 2021-07-21:
刷了docker版本 还是不到一天会重启

45142918, 2021-07-21:
> 刷了docker版本 还是不到一天会重启

也是凌晨4点？

mouse0907, 2021-07-21:
我现在刷的是0718版本，变成了2天关机一次~我在尝试是不是卡的问题。好揪心~

NathanDai, 2021-07-21:
> > 刷了docker版本 还是不到一天会重启
> 
> 也是凌晨4点？

不是 一次是早上五点 一次是下午16点

DHDAXCW, 2021-07-22:
我刚才看了另一个群有人经常频繁重启，换了新r4s就好了。。。

NathanDai, 2021-07-22:
> 我刚才看了另一个群有人经常频繁重启，换了新r4s就好了。。。

我把docker  禁用了 目前正常运行了一天

mouse0907, 2021-07-22:
> 我刚好又一个群有人轻松重启，换了新的r4s就好了。。。

我们这种情况不是设备重启，而是到时间就自动关机

DHDAXCW, 2021-07-22:
我今天卖了r4s 4G的，出200元卖掉了。手里还有好几个r4s

NathanDai, 2021-07-24:
> > 我刚好又一个群有人轻松重启，换了新的r4s就好了。。。
> 
> 我们这种情况不是设备重启，而是到时间就自动关机
> 我不是说重启存在重启，而是死机，频繁死机。还有你跑docker，该不会是把内存和TF跑累了吧？

有可能 我看了下重启前的系统日志

Sat Jul 24 08:50:31 2021 kern.alert kernel: [57566.888059] Unable to handle kernel paging request at virtual address >
Sat Jul 24 08:50:31 2021 kern.alert kernel: [57566.888791] Mem abort info:
Sat Jul 24 08:50:31 2021 kern.alert kernel: [57566.889058]   ESR = 0x96000047
Sat Jul 24 08:50:31 2021 kern.alert kernel: [57566.889342]   EC = 0x25: DABT (current EL), IL = 32 bits
Sat Jul 24 08:50:31 2021 kern.alert kernel: [57566.889830]   SET = 0, FnV = 0
Sat Jul 24 08:50:31 2021 kern.alert kernel: [57566.890122]   EA = 0, S1PTW = 0
Sat Jul 24 08:50:31 2021 kern.alert kernel: [57566.890419] Data abort info:
Sat Jul 24 08:50:31 2021 kern.alert kernel: [57566.890697]   ISV = 0, ISS = 0x00000047
Sat Jul 24 08:50:31 2021 kern.alert kernel: [57566.891064]   CM = 0, WnR = 1
Sat Jul 24 08:50:31 2021 kern.alert kernel: [57566.891357] swapper pgtable: 4k pages, 48-bit VAs, pgdp=0000000002a880>
Sat Jul 24 08:50:31 2021 kern.alert kernel: [57566.891987] [ffff800010b2e520] pgd=00000000f77ff003, pud=00000000f77fe                                                                                                                        >
Sat Jul 24 08:50:31 2021 kern.emerg kernel: [57566.892947] Internal error: Oops: 96000047 [#1] PREEMPT SMP
好像是内存的问题

DHDAXCW, 2021-07-30:
这似乎关系不大.我前几天做了测试,也是一样日志,没死机.最好就是换个新机器解愁

45142918, 2021-07-30:
> 这似乎关系不大.我前几天做了测试,也是一样日志,没死机.最好就是换个新机器解愁

已放弃这个固件

DHDAXCW, 2021-08-01:
好的

closed 2021-08-01 by DHDAXCW