DHDAXCW/NanoPi-R4S-R4SE #90: opkg update 报Signature check failed.

vichbb, 2021-07-18:
[image]

DHDAXCW, 2021-07-18:
啊这...你有没有更好的源？

DHDAXCW, 2021-07-18:
这个我晚上解决一下哈

DHDAXCW, 2021-07-18:
这个同步到一半，没同步完可能，大概率是腾讯云的问题

vichbb, 2021-07-18:
我这边换了清华也不行，注释：#option check_signature，就可以了

vichbb, 2021-07-18:
那麻烦你晚点再看一下吧。

DHDAXCW, 2021-07-18:
ojbk

vichbb, 2021-07-18:
[image]

vichbb, 2021-07-18:
[Link](https://www.cnblogs.com/oneseven/p/8974730.html)

DHDAXCW, 2021-07-20:
这是两个解决方法

1. 改回之前用snapshot源，但snapshot源的软件不一定稳定，我个人不太推荐。

2. 把OpenWRT 18.06的公钥加进去

wget 'https://git.openwrt.org/?p=keyring.git;a=blob_plain;f=usign/1035ac73cc4e59e3' -O 1035ac73cc4e59e3
opkg-key add 1035ac73cc4e59e3

就搞定。验证的方法是看看 /etc/opkg/keys 里面有没有 1035ac73cc4e59e3 这个文件

因为固件里面，/etc/opkg/keys  里面只存了两个Keys：

a551f22392e57422   这是友善自己的公钥
b5043e70f9a75cde   这是snapshot源的公钥

https://openwrt.org/docs/guide-user/security/signatures

DHDAXCW, 2021-07-27:
已解决，明天版本 
[image]