DHDAXCW/NanoPi-R4S-R4SE #110: R4S正式版lan口不定时掉线，刚刚开始玩openwrt，不清楚原因o(╥﹏╥)o

lihuaye, 2021-09-08:
Tue Sep  7 19:21:05 2021 kern.info kernel: [   31.151168] vbus_typec: disabling
Tue Sep  7 19:21:07 2021 daemon.info procd: Instance dnsmasq::cfg01411c pid 3771 not stopped on SIGTERM, sending SIGKILL instead
Tue Sep  7 19:21:07 2021 daemon.info dnsmasq[6297]: Connected to system UBus
Tue Sep  7 19:21:11 2021 daemon.notice procd: /etc/rc.d/S99openclash: OpenClash Already Start!
Tue Sep  7 19:21:11 2021 daemon.info qbittorrent: Instance "main" disabled.
Tue Sep  7 19:21:11 2021 daemon.notice procd: /etc/rc.d/S99qbittorrent: Command failed: Invalid argument
Tue Sep  7 19:21:11 2021 cron.err crond[6514]: crond (busybox 1.33.1) started, log level 5
Tue Sep  7 19:21:11 2021 daemon.notice ttyd[6572]: ttyd 1.6.3 (libwebsockets 4.1.6-20200915-799-gd7a7821cc)
Tue Sep  7 19:21:11 2021 daemon.notice ttyd[6572]: tty configuration:
Tue Sep  7 19:21:11 2021 daemon.notice ttyd[6572]:   start command: /bin/login
Tue Sep  7 19:21:11 2021 daemon.notice ttyd[6572]:   close signal: SIGHUP (1)
Tue Sep  7 19:21:11 2021 daemon.notice ttyd[6572]:   terminal type: xterm-256color
Tue Sep  7 19:21:11 2021 daemon.notice ttyd[6572]: LWS: 4.1.6-20200915-799-gd7a7821cc, loglevel 7
Tue Sep  7 19:21:11 2021 daemon.notice ttyd[6572]: NET CLI SRV H1 H2 WS IPV6-off
Tue Sep  7 19:21:11 2021 daemon.notice ttyd[6572]:    /usr/lib/libwebsockets-evlib_uv.so
Tue Sep  7 19:21:11 2021 daemon.notice ttyd[6572]:  Using foreign event loop...
Tue Sep  7 19:21:11 2021 daemon.notice ttyd[6572]:  Listening on port: 7681
Tue Sep  7 19:21:11 2021 daemon.notice procd: /etc/rc.d/S99zerotier: disabled in config
Tue Sep  7 19:21:11 2021 daemon.info procd: - init complete -
Tue Sep  7 19:21:16 2021 daemon.info procd: Instance haproxy::instance1 s in a crash loop 6 crashes, 0 seconds since last crash
Tue Sep  7 19:21:16 2021 user.notice firewall: Reloading firewall due to ifup of wan (pppoe-wan)
Tue Sep  7 19:21:20 2021 daemon.info dnsmasq[7465]: Connected to system UBus
Tue Sep  7 19:22:34 2021 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Tue Sep  7 19:22:34 2021 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Tue Sep  7 19:22:38 2021 daemon.info dnsmasq[8380]: Connected to system UBus
Tue Sep  7 19:23:17 2021 daemon.notice miniupnpd[5700]: shutting down MiniUPnPd
Tue Sep  7 19:23:17 2021 daemon.notice miniupnpd[8639]: HTTP listening on port 5000
Tue Sep  7 19:23:17 2021 daemon.notice miniupnpd[8639]: Listening for NAT-PMP/PCP traffic on port 5351
Tue Sep  7 19:53:23 2021 daemon.info netdata[4769]: Initializing file /var/cache/netdata/cpu.cpu0_softirqs/HRTIMER.db.
Tue Sep  7 19:53:23 2021 daemon.info netdata[4769]: Initializing file /var/cache/netdata/cpu.cpu1_softirqs/HRTIMER.db.
Tue Sep  7 20:01:51 2021 daemon.info netdata[4769]: Initializing file /var/cache/netdata/cpu.cpu2_softirqs/HRTIMER.db.
Tue Sep  7 20:40:37 2021 daemon.info netdata[4769]: Initializing file /var/cache/netdata/cpu.cpu3_softirqs/HRTIMER.db.
Tue Sep  7 21:15:13 2021 daemon.info netdata[4769]: Initializing file /var/cache/netdata/cpu.cpu5_softirqs/HRTIMER.db.
Tue Sep  7 23:25:45 2021 kern.info kernel: [14711.515458] r8169 0000:01:00.0 eth1: Link is Down
Tue Sep  7 23:25:45 2021 daemon.notice netifd: Network device 'eth1' link is down
Tue Sep  7 23:25:45 2021 kern.info kernel: [14711.516891] br-lan: port 1(eth1) entered disabled state
Tue Sep  7 23:25:46 2021 daemon.notice netifd: bridge 'br-lan' link is down
Tue Sep  7 23:25:46 2021 daemon.notice netifd: Interface 'lan' has link connectivity loss
Tue Sep  7 23:25:48 2021 daemon.notice netifd: Network device 'eth1' link is up
Tue Sep  7 23:25:48 2021 daemon.notice netifd: bridge 'br-lan' link is up
Tue Sep  7 23:25:48 2021 daemon.notice netifd: Interface 'lan' has link connectivity
Tue Sep  7 23:25:48 2021 kern.info kernel: [14714.722982] r8169 0000:01:00.0 eth1: Link is Up - 1Gbps/Full - flow control off
Tue Sep  7 23:25:48 2021 kern.info kernel: [14714.723701] br-lan: port 1(eth1) entered blocking state
Tue Sep  7 23:25:48 2021 kern.info kernel: [14714.724167] br-lan: port 1(eth1) entered forwarding state
Tue Sep  7 23:26:00 2021 daemon.notice netifd: Network device 'eth1' link is down
Tue Sep  7 23:26:00 2021 kern.info kernel: [14726.229356] r8169 0000:01:00.0 eth1: Link is Down
Tue Sep  7 23:26:00 2021 kern.info kernel: [14726.229989] br-lan: port 1(eth1) entered disabled state
Tue Sep  7 23:26:01 2021 daemon.notice netifd: bridge 'br-lan' link is down
Tue Sep  7 23:26:01 2021 daemon.notice netifd: Interface 'lan' has link connectivity loss
Tue Sep  7 23:26:11 2021 daemon.notice netifd: Network device 'eth1' link is up
Tue Sep  7 23:26:11 2021 daemon.notice netifd: bridge 'br-lan' link is up
Tue Sep  7 23:26:11 2021 daemon.notice netifd: Interface 'lan' has link connectivity
Tue Sep  7 23:26:11 2021 kern.info kernel: [14737.568159] r8169 0000:01:00.0 eth1: Link is Up - 1Gbps/Full - flow control off
Tue Sep  7 23:26:11 2021 kern.info kernel: [14737.569005] br-lan: port 1(eth1) entered blocking state
Tue Sep  7 23:26:11 2021 kern.info kernel: [14737.569472] br-lan: port 1(eth1) entered forwarding state
Tue Sep  7 23:26:11 2021 kern.info kernel: [14737.916792] r8169 0000:01:00.0 eth1: Link is Down
Tue Sep  7 23:26:12 2021 daemon.notice netifd: Network device 'eth1' link is down
Tue Sep  7 23:26:12 2021 kern.info kernel: [14738.629228] br-lan: port 1(eth1) entered disabled state
Tue Sep  7 23:26:13 2021 daemon.notice netifd: bridge 'br-lan' link is down
Tue Sep  7 23:26:13 2021 daemon.notice netifd: Interface 'lan' has link connectivity loss
Tue Sep  7 23:26:13 2021 daemon.info netdata[4769]: Initializing file /var/cache/netdata/net_errors.eth1/main.db.
Tue Sep  7 23:26:13 2021 daemon.info netdata[4769]: Initializing file /var/cache/netdata/net_errors.eth1/inbound.db.
Tue Sep  7 23:26:13 2021 daemon.info netdata[4769]: Initializing file /var/cache/netdata/net_errors.eth1/outbound.db.
Tue Sep  7 23:26:23 2021 daemon.notice netifd: Network device 'eth1' link is up
Tue Sep  7 23:26:23 2021 daemon.notice netifd: bridge 'br-lan' link is up
Tue Sep  7 23:26:23 2021 daemon.notice netifd: Interface 'lan' has link connectivity
Tue Sep  7 23:26:23 2021 kern.info kernel: [14749.185135] r8169 0000:01:00.0 eth1: Link is Up - 1Gbps/Full - flow control off
Tue Sep  7 23:26:23 2021 kern.info kernel: [14749.185853] br-lan: port 1(eth1) entered blocking state
Tue Sep  7 23:26:23 2021 kern.info kernel: [14749.186319] br-lan: port 1(eth1) entered forwarding state
Wed Sep  8 00:00:00 2021 cron.err crond[6514]: USER root pid 23066 cmd /usr/share/openclash/openclash.sh
[... 63 lines cut ...]
Wed Sep  8 10:07:05 2021 kern.info kernel: [53192.378871] br-lan: port 1(eth1) entered disabled state
Wed Sep  8 10:07:07 2021 daemon.notice netifd: bridge 'br-lan' link is down
Wed Sep  8 10:07:07 2021 daemon.notice netifd: Interface 'lan' has link connectivity loss
Wed Sep  8 10:07:09 2021 daemon.notice netifd: Network device 'eth1' link is up
Wed Sep  8 10:07:09 2021 daemon.notice netifd: bridge 'br-lan' link is up
Wed Sep  8 10:07:09 2021 daemon.notice netifd: Interface 'lan' has link connectivity
Wed Sep  8 10:07:09 2021 kern.info kernel: [53195.577324] r8169 0000:01:00.0 eth1: Link is Up - 1Gbps/Full - flow control off
Wed Sep  8 10:07:09 2021 kern.info kernel: [53195.578045] br-lan: port 1(eth1) entered blocking state
Wed Sep  8 10:07:09 2021 kern.info kernel: [53195.578512] br-lan: port 1(eth1) entered forwarding state
Wed Sep  8 10:19:20 2021 kern.info kernel: [53926.824975] TCP: request_sock_TCP: Possible SYN flooding on port 53. Sending cookies.  Check SNMP counters.
Wed Sep  8 10:19:47 2021 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Wed Sep  8 10:19:47 2021 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Wed Sep  8 10:19:51 2021 daemon.info dnsmasq[23413]: Connected to system UBus
Wed Sep  8 10:19:51 2021 cron.err crond[23477]: crond (busybox 1.33.1) started, log level 5
Wed Sep  8 10:19:52 2021 kern.info kernel: [53958.558891] TCP: request_sock_TCP: Possible SYN flooding on port 53. Sending cookies.  Check SNMP counters.
Wed Sep  8 10:20:08 2021 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Wed Sep  8 10:20:08 2021 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Wed Sep  8 10:20:11 2021 daemon.info dnsmasq[25078]: Connected to system UBus
[... cut ...]
Wed Sep  8 10:26:04 2021 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Wed Sep  8 10:26:04 2021 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Wed Sep  8 10:26:08 2021 daemon.info dnsmasq[12083]: Connected to system UBus
Wed Sep  8 10:26:10 2021 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Wed Sep  8 10:26:10 2021 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Wed Sep  8 10:26:13 2021 daemon.info dnsmasq[12736]: Connected to system UBus

DHDAXCW, 2021-09-11:
flow control问题，去把Shortcut-FE关闭再试，在acc加速模式里

lihuaye, 2021-09-14:
> flow control问题，去把Shortcut-FE关闭再试，在acc加速模式里

以为是机子的问题，拿去换新了。。。回来了我再试下，感谢Thanks♪(･ω･)ﾉ

lihuaye, 2021-09-22:
> flow control问题，去把Shortcut-FE关闭再试，在acc加速模式里

经过几天测试，关了Shortcut-FE就没有在掉lan口了，谢谢

closed 2021-09-23 by lihuaye