DHDAXCW/NanoPi-R4S-R4SE #137: 无法自动更新

BuBuxsy, 2021-11-26:
wget https://raw.githubusercontent.com/lone-wind/scripts/main/onlineupdate.sh && sh onlineupdate.sh

--2021-11-27 04:58:34--  https://raw.githubusercontent.com/lone-wind/scripts/main/onlineupdate.sh
Resolving raw.githubusercontent.com... 0.0.0.0
Connecting to raw.githubusercontent.com|0.0.0.0|:443... connected.
ERROR: cannot verify raw.githubusercontent.com's certificate, issued by 'CN=OpenWrt,O=OpenWrta0c96672,L=Unknown,ST=Somewhere,C=ZZ':
  Self-signed certificate encountered.
    ERROR: certificate common name 'OpenWrt' doesn't match requested host name 'raw.githubusercontent.com'.
To connect to raw.githubusercontent.com insecurely, use `--no-check-certificate'.

BuBuxsy, 2021-11-26:
$ wget --no-check-certificate  https://raw.githubusercontent.com/lone-wind/scripts/main/onlineupdate.sh

--2021-11-27 05:02:02--  https://raw.githubusercontent.com/lone-wind/scripts/main/onlineupdate.sh
Resolving raw.githubusercontent.com... 0.0.0.0
Connecting to raw.githubusercontent.com|0.0.0.0|:443... connected.
WARNING: cannot verify raw.githubusercontent.com's certificate, issued by 'CN=OpenWrt,O=OpenWrta0c96672,L=Unknown,ST=Somewhere,C=ZZ':
  Self-signed certificate encountered.
    WARNING: certificate common name 'OpenWrt' doesn't match requested host name 'raw.githubusercontent.com'.
HTTP request sent, awaiting response... 404 Not Found
2021-11-27 05:02:02 ERROR 404: Not Found.

这样子也没用

DHDAXCW, 2021-11-27:
现在不合适了，等上游帮忙改改

BuBuxsy, 2021-11-27:
谢谢骷髅大

closed 2021-11-27 by BuBuxsy