DHDAXCW/NanoPi-R4S-R4SE #139: 使用openclash时，无论openclash工作在何种模式下，每半小时都会提示“重置 OpenClash 防火墙规则...”，并会导致游戏、视频等连接掉线。

cjxzdzh, 2021-12-04:
使用openclash时，无论openclash工作在何种模式下，每半小时都会提示“重置 OpenClash 防火墙规则...”，并会导致游戏、视频等连接掉线。
2021-12-04 14:11:49 重置 OpenClash 防火墙规则...
2021-12-04 13:41:48 重置 OpenClash 防火墙规则...
2021-12-04 13:11:43 重置 OpenClash 防火墙规则...
2021-12-04 12:41:43 重置 OpenClash 防火墙规则...
2021-12-04 12:11:41 重置 OpenClash 防火墙规则...
2021-12-04 11:41:38 重置 OpenClash 防火墙规则...
2021-12-04 11:11:36 重置 OpenClash 防火墙规则...
2021-12-04 10:41:32 重置 OpenClash 防火墙规则...
2021-12-04 10:11:31 重置 OpenClash 防火墙规则...
2021-12-04 09:41:29 重置 OpenClash 防火墙规则...
2021-12-04 09:17:57 第十步: 添加计划任务,启动进程守护程序...
2021-12-04 09:17:56 第九步: 重启 Dnsmasq 程序...
2021-12-04 09:17:52 第八步: 设置防火墙规则...
2021-12-04 09:17:52 第七步: 设置控制面板...
2021-12-04 09:17:52 第六步: 等待主程序下载外部文件...
2021-12-04 09:17:49 第五步: 检查内核启动状态...
2021-12-04 09:17:49 提示: 未检测到特殊配置，调用 Dev 内核启动...
2021-12-04 09:17:49 第四步: 启动主程序...
2021-12-04 09:17:46 第三步: 修改配置文件...
2021-12-04 09:17:46 提示: 由于文件【 /etc/config/openclash 】被修改，暂停快速启动...
2021-12-04 09:17:45 第二步: 组件运行前检查...
2021-12-04 09:17:45 第一步: 获取配置...
2021-12-04 09:17:45 OpenClash 开始启动...
2021-12-04 09:17:45 第六步：删除 OpenClash 残留文件...
2021-12-04 09:17:45 第五步: 重启 Dnsmasq 程序...
2021-12-04 09:17:45 第四步: 关闭 Clash 主程序...
2021-12-04 09:17:45 第三步: 关闭 OpenClash 守护程序...
2021-12-04 09:17:43 第二步: 删除 OpenClash 防火墙规则...
2021-12-04 09:17:43 第一步: 备份当前策略组状态...

oniLand, 2021-12-15:
[image]
是否开启了自动重启？

cjxzdzh, 2021-12-15:
并未开启openclash的定时重启。即使使用Passwall，也依然能从passwall的日志中观察到每半小时会自动重启passwall的日志信息。另外在crontab表内，也未能找到和半小时相关的功能。

closed 2021-12-21 by DHDAXCW

cjxzdzh, 2022-01-23:
仔细检查了下，发现WAN口启用了IPV6后，并且拨号后能工作在公网双栈模式下，就会导致openclash报“重置 OpenClash 防火墙规则”的问题。禁用后就一切正常了。
可惜了ipv6