DHDAXCW/NanoPi-R4S-R4SE #142: RTL8814AU for NetGear A7000

HonorWater, 2021-12-07:
在之前的ticket里面看到有人提过这个需求，已经被close了，但是我测试了几个最新的固件，都没法驱动，帮忙再添加一下。
设备：R4S
谢谢。

DHDAXCW, 2021-12-07:
RTL8814AU这种网卡很新，我项目里没有这个驱动

DHDAXCW, 2021-12-07:
已添加，等明天

HonorWater, 2021-12-08:
多谢，我明天测试一下再来反馈。

DHDAXCW, 2021-12-08:
今天可以试试了

HonorWater, 2021-12-08:
已经测试过了，可以直接驱动使用了，但是测试下来不知是驱动的问题还是其他问题，链接速度从866Mbps经过一段时间后就降为3Mbps，然后网络就不通了。

内核日志：

[   20.472647] br-lan: port 2(wlan0) entered blocking state
[   20.473126] br-lan: port 2(wlan0) entered disabled state
[   20.473914] device wlan0 entered promiscuous mode
[   20.474745] br-lan: port 2(wlan0) entered blocking state
[   20.475218] br-lan: port 2(wlan0) entered forwarding state
[   20.476075] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[   22.069693] r8169 0000:01:00.0 eth1: Link is Up - 1Gbps/Full - flow control off
[   22.070401] br-lan: port 1(eth1) entered blocking state
[   22.070868] br-lan: port 1(eth1) entered forwarding state
[   39.532874] vbus_typec: disabling
[10171.422261] xhci-hcd xhci-hcd.0.auto: xHCI host not responding to stop endpoint command.
[10171.436570] xhci-hcd xhci-hcd.0.auto: xHCI host controller not responding, assume dead
[10171.437393] xhci-hcd xhci-hcd.0.auto: HC died; cleaning up
[10171.438029] usb 6-1: USB disconnect, device number 2
[10171.441382] br-lan: port 2(wlan0) entered disabled state
[10171.446835] device wlan0 left promiscuous mode
[10171.447239] br-lan: port 2(wlan0) entered disabled state

HonorWater, 2021-12-09:
配置上应该没什么问题，之前另外一款A6210一样的配置，也很稳定。
现在不能确定的是驱动问题还是我台A7000设备问题，或者R4S的本身3A供电问题等等。
上次也有一个朋友求这个驱动，不知他能否看到信息，帮忙一起测试验证一下。

DHDAXCW, 2021-12-09:
A6210这种网卡是没问题的，很稳，至于那个螃蟹的，头疼。关键我还没有那些几个网卡

HonorWater, 2021-12-10:
辛苦了！
这个A7000也是头脑一热就追了一个，而且块头比A6210大多了。

closed 2021-12-21 by DHDAXCW