DHDAXCW/NanoPi-R4S-R4SE #145: 大佬，能否帮忙加下TL-WDN7201H驱动，RTL8814AU芯片

gnastnosaj, 2021-12-23:
大佬，能否帮忙加下TL-WDN7201H驱动，RTL8814AU芯片
[TL-WDN7201H V1.0 Linux驱动程序.zip](https://github.com/DHDAXCW/NanoPi-R4S/files/7740534/TL-WDN7201H.V1.0.Linux.zip)

_Originally posted by @gnastnosaj in https://github.com/DHDAXCW/NanoPi-R4S/issues/6#issuecomment-997357458_

DHDAXCW, 2021-12-25:
我用rtl8812au驱动，但是里面包含了rtl8814au驱动，可以识别的 https://github.com/baxobox/add_rtl8812ac/blob/main/rtl8812au-ac/Makefile#L25

gnastnosaj, 2021-12-26:
呃，我看了那个项目说是不再支持rtl8814au，作者另外开了一个rtl8814的单独项目，我先自己加上去试试

gnastnosaj, 2021-12-31:
@DHDAXCW 大佬，github action还能跑完么，我就成功一次，最近都是超时结束。

DHDAXCW, 2021-12-31:
那我重新加吧

DHDAXCW, 2021-12-31:
我给你pr了

junjiexia1987, 2021-12-31:
大神，RTL8812bu驱动有吗？

gnastnosaj, 2022-01-03:
😄搞定了，之前没弄过驱动，这次手贱买了个比较新的usb网卡，原来要把对应的设备id加到驱动里才会认，对应的源码是这个文件os_dep/linux/usb_intf.c

closed 2022-01-03 by gnastnosaj

HonorWater, 2022-01-05:
@gnastnosaj 请问是自己编译之后解决的吗？我手头的A7000也是这个新品，用的这里的固件，能识别，但是极其不稳定。

gnastnosaj, 2022-01-12:
@HonorWater 试试这个驱动
[rtl8814au.zip](https://github.com/DHDAXCW/NanoPi-R4S/files/7852272/rtl8814au.zip)

gnastnosaj, 2022-01-12:
cp rtl8814au.ko /lib/modules/5.4.162/
rmmod rtl8812au
modprobe rtl8814au