DHDAXCW/NanoPi-R4S-R4SE #147: R4S的固件_DOCKER全插件版本.你们是否用久了会出现LAN口从千兆变成速度只有250左右.

showme2005, 2021-12-27:
我是R4S的1G版本.固件是2021.11.18号的固件版本..我现在外挂了一块QB下载硬盘..最开始还算正常.但是过了几天.我的硬盘速度从100MB/秒变成只有5MB/秒.而且LAN口的速度用iperf3测试也从940MB/秒变成了250MB左右..不知道是什么原因...重启之后这些都恢复正常...

hosence, 2021-12-27:
我的是R4S的4G版本，固件是20211220正式版，正常宽带能跑满500M，但是开机一天半左右，网速就会下降，前两天掉到80多M，昨天上午重启恢复正常。今天晚上（开机1天8小时）掉到150兆左右，重启就恢复了。后台只开了HelloWorld，sfe和TurboACC里面全关了，smartDNS一直开着。但我跟你不一样的是我iperf测速没问题

showme2005, 2021-12-28:
> 我的是R4S的4G版本，固件是20211220正式版，正常宽带能跑满500M，但是开机一天半左右，网速就会下降，前两天掉到80多M，昨天上午重启恢复正常。今天晚上（开机1天8小时）掉到150兆左右，重启就恢复了。后台只开了HelloWorld，sfe和TurboACC里面全关了，smartDNS一直开着。但我跟你不一样的是我iperf测速没问题

我只开了passwall smartdns adguardhome 还有一个docker

hosence, 2021-12-29:
> > 我的是R4S的4G版本，固件是20211220正式版，正常宽带能跑满500M，但是开机一天半左右，网速就会下降，前两天掉到80多M，昨天上午重启恢复正常。今天晚上（开机1天8小时）掉到150兆左右，重启就恢复了。后台只开了HelloWorld，sfe和TurboACC里面全关了，smartDNS一直开着。但我跟你不一样的是我iperf测速没问题
> 
> 我只开了passwall smartdns adguardhome 还有一个docker

我现在换了bleach的固件，现在开机还没到一天，等过几个小时我再试试，验证一下到底是固件问题还是硬件问题。

hosence, 2021-12-29:
> > > 我的是R4S的4G版本，固件是20211220正式版，正常宽带能跑满500M，但是开机一天半左右，网速就会下降，前两天掉到80多M，昨天上午重启恢复正常。今天晚上（开机1天8小时）掉到150兆左右，重启就恢复了。后台只开了HelloWorld，sfe和TurboACC里面全关了，smartDNS一直开着。但我跟你不一样的是我iperf测速没问题
> > 
> > 
> > 我只开了passwall smartdns adguardhome 还有一个docker
> 
> 我现在换了bleach的固件，现在开机还没到一天，等过几个小时我再试试，验证一下到底是固件问题还是硬件问题。

目前开机1d7h，没有发现问题
有空刷回骷髅头老版本再试试。
刷了好多固件，我自己还是比较喜欢骷髅头的，界面风格好看，插件丰富。

Sorkai, 2022-01-08:
我也遇到了，使用十多个小时后开始丢包掉速，确定散热没有问题（大风扇直吹表面冰冰凉），重启可以恢复，感觉可能是PDNSD的问题，不过我有两个r4s，一会我换一个试试，排除一下硬件问题

Sorkai, 2022-01-08:
> 我的是R4S的4G版本，固件是20211220正式版，正常宽带能跑满500M，但是开机一天半左右，网速就会下降，前两天掉到80多M，昨天上午重启恢复正常。今天晚上（开机1天8小时）掉到150兆左右，重启就恢复了。后台只开了HelloWorld，sfe和TurboACC里面全关了，smartDNS一直开着。但我跟你不一样的是我iperf测速没问题

我开的也是HelloWorld，千兆宽带，和你问题差不多，感觉是HelloWorld里的PDNSD的问题，还有就是我把TurboACC里的PDNSD关了以后问题改善了很多，可HelloWorld里只能用PDNSD

hosence, 2022-01-08:
> > 我的是R4S的4G版本，固件是20211220正式版，正常宽带能跑满500M，但是开机一天半左右，网速就会下降，前两天掉到80多M，昨天上午重启恢复正常。今天晚上（开机1天8小时）掉到150兆左右，重启就恢复了。后台只开了HelloWorld，sfe和TurboACC里面全关了，smartDNS一直开着。但我跟你不一样的是我iperf测速没问题
> 
> 我开的也是HelloWorld，千兆宽带，和你问题差不多，感觉是HelloWorld里的PDNSD的问题，还有就是我把TurboACC里的PDNSD关了以后问题改善了很多，可HelloWorld里只能用PDNSD

目前刷了20220103的正式版，稳定运行5天了，没有掉速。20220101的也不会掉速。但是群里有人说他刷的20220101版本掉速。这到底是哪的问题我也没弄明白。

Sorkai, 2022-01-08:
> 目前刷了20220103的正式版，稳定运行5天了，没有掉速。20220101的也不会掉速。但是群里有人说他刷的20220101版本掉速。这到底是哪的问题我也没弄明白。

01的我试了，好像有改善？？但不完全好，现在正在用今天(20220108)的固件，上午刷的，现在还没出问题呢，多用两天看看吧

Sorkai, 2022-01-09:
我试了下，延时好像有改善，有丢包，速度也有明显下降

huluobo12138, 2022-01-15:
> 

我也遇到了同样的问题，千兆掉速到200兆，手机连接无线网络只有几十兆，仅仅就是开了HelloWorld

Sorkai, 2022-01-15:
我要收回我之前说的话，，更新了也没改善😂

hosence, 2022-01-15:
> 

刷20220103试试，我还是觉得这个稳

Sorkai, 2022-01-15:
试过了，感觉也没啥区别

hosence, 2022-01-16:
> 试过了，感觉也没啥区别

那就再换个固件吧，2021.12.12和2022.1.3这两个我用着没啥问题。

fayeyuan, 2022-01-17:
我也遇到了同样问题2021.11.12

Sorkai, 2022-01-18:
我怀疑是系统中有漏洞被利用，或是被攻击，刚才特别卡，网特别慢，我把所有连接导出拉了个Excel表，统计了下发现有五千多个目标地址是xx.xx.xx.xx:1900的UDP连接，xx.xx.xx.xx是我当前的公网ip(我这里是直接给公网的)，我用 iptables丢弃了所有端口是1900的UDP连接，问题好转了90%，速度快了好多，延时问题解决了但没完全解决(比正常高出几十毫秒)

还发现了一些其他的：
1.出现问题时CPU占用最高的是ksoftirqd(网络中断相关)
2.丢弃1900端口所有连接后ksoftirqd进程占用有下降，但依旧不正常
3.整体CPU占用最多的是nice类进程，也就是改变过优先级的进程，并且硬中断占用不正常
4.top命令显示的cpu占用总和相加距实际总占用有很大总差距，怀疑有进程被隐藏？？？？

这里希望大家再遇到降速问题时检查下列问题，看看咱们的问题有没有什么共同点，谢谢💕
1.在`状态`-`概览`页查看**总连接数**是否不正常
2.思考自己的路由器是否能获得**公网IP**
3.在`状态`-`实时信息`-`连接`中看看是不是有很多UDP连接，有能力的把下方连接信息复制下来到Excel中排序分类汇总下，看看有没有很多目标地址端口是1900的连接
4.在`系统`-`TTYD终端`中输入`top`命令查看当前ksoftirqd是否高占用
5.在`网络`-`防火墙`-`自定义规则`中添加`iptables -I FORWARD -p udp --dport 1900 -j DROP`规则，并点击**重启防火墙**，等待1分钟，检查问题是否有改善

canoziia, 2022-01-19:
> 我怀疑是系统中有漏洞被利用，或是被攻击，刚才特别卡，网特别慢，我把所有连接导出拉了个Excel表，统计了下发现有五千多个目标地址是xx.xx.xx.xx:1900的UDP连接，xx.xx.xx.xx是我当前的公网ip(我这里是直接给公网的)，我用 iptables丢弃了所有端口是1900的UDP连接，问题好转了90%，速度快了好多，延时问题解决了但没完全解决(比正常高出几十毫秒) [...]

[image]
占用端口的应该是这个linkease

showme2005, 2022-01-26:
我怀疑是adguardhome 的问题...当然我没有测试骷髅大神的固件 ,我自己编译的固件之前也偶到过同样的问题..没有找到原因...但是..某一次刷固件的过程中.我少配置了adguardhome 这个插件,,,用到现在 居然没有出现LAN口降速和死机情况....现在运行的 插件 只有smartdns+passwall+docker

Nxir, 2022-01-30:
20220103无docker正式版  同样遇到掉速问题   重启后正常   服务只启动了openclash

Nxir, 2022-01-31:
> 20220103无docker正式版 同样遇到掉速问题 重启后正常 服务只启动了openclash

我把网络共享samba关掉后(系统->启动项->samba->禁用)  暂时看来情况有好转  之前几个小时就会掉速  现在半天了还没掉  具体原因不清楚  再观察观察

另外 大多数时间我的在线用户数在35-40之间  我留意到 samba关闭之前活动连接数一直有两三千  关闭后就维持在一千多

canoziia, 2022-02-02:
今天又遇到严重的掉速问题，管理页面重新连接wan口获得新的公网ip后，情况立刻消失。掉速严重的时候可以看到有6000多udp连接，极不正常。

canoziia, 2022-02-02:
截取了一段连接列表:
<details>
<summary>展开</summary>
<pre>
```sh
# root @ FusionWrt in ~ [20:52:37]
$  netstat -antulp
Active Internet connections (servers and established)
Proto Recv-Q Send-Q Local Address           Foreign Address         State       PID/Program name
tcp        0      0 0.0.0.0:1688            0.0.0.0:*               LISTEN      4751/vlmcsd
tcp        0      0 0.0.0.0:443             0.0.0.0:*               LISTEN      3217/uhttpd
tcp        0      0 127.0.0.1:8125          0.0.0.0:*               LISTEN      9654/netdata
tcp        0      0 127.0.0.1:445           0.0.0.0:*               LISTEN      4149/smbd
tcp        0      0 192.168.2.1:445         0.0.0.0:*               LISTEN      4149/smbd
tcp        0      0 0.0.0.0:19999           0.0.0.0:*               LISTEN      9654/netdata
tcp        0      0 192.168.2.1:7681        0.0.0.0:*               LISTEN      10524/ttyd
tcp        0      0 0.0.0.0:2049            0.0.0.0:*               LISTEN      -
tcp        0      0 0.0.0.0:1026            0.0.0.0:*               LISTEN      2637/php8-fcgi
tcp        0      0 0.0.0.0:5000            0.0.0.0:*               LISTEN      13772/miniupnpd
tcp        0      0 0.0.0.0:32777           0.0.0.0:*               LISTEN      -
tcp        0      0 0.0.0.0:873             0.0.0.0:*               LISTEN      4951/rsync
tcp        0      0 0.0.0.0:32778           0.0.0.0:*               LISTEN      9867/rpc.statd
tcp        0      0 0.0.0.0:5355            0.0.0.0:*               LISTEN      10924/wsdd2
tcp        0      0 127.0.0.1:139           0.0.0.0:*               LISTEN      4149/smbd
tcp        0      0 192.168.2.1:139         0.0.0.0:*               LISTEN      4149/smbd
tcp        0      0 0.0.0.0:32780           0.0.0.0:*               LISTEN      9868/rpc.mountd
tcp        0      0 0.0.0.0:111             0.0.0.0:*               LISTEN      2060/rpcbind
tcp        0      0 0.0.0.0:80              0.0.0.0:*               LISTEN      3217/uhttpd
tcp        0      0 115.221.64.114:53       0.0.0.0:*               LISTEN      15989/dnsmasq
tcp        0      0 127.0.0.1:53            0.0.0.0:*               LISTEN      15989/dnsmasq
tcp        0      0 192.168.2.1:53          0.0.0.0:*               LISTEN      15989/dnsmasq
tcp        0      0 0.0.0.0:21              0.0.0.0:*               LISTEN      3320/vsftpd
tcp        0      0 0.0.0.0:3702            0.0.0.0:*               LISTEN      10924/wsdd2
tcp        0      0 192.168.2.1:22          0.0.0.0:*               LISTEN      14229/dropbear
tcp        0      0 115.221.64.114:57110    101.230.149.37:11565    TIME_WAIT   -
tcp        0      0 115.221.64.114:56874    101.230.149.37:11565    TIME_WAIT   -
tcp        0      0 115.221.64.114:57212    101.230.149.37:11565    ESTABLISHED 1107/clash
tcp        0      0 192.168.2.1:22          192.168.2.237:58297     ESTABLISHED 19034/dropbear
tcp        0      0 115.221.64.114:58610    101.230.149.37:11565    ESTABLISHED 1107/clash
tcp        0      0 115.221.64.114:57870    101.230.149.37:11565    ESTABLISHED 1107/clash
tcp        0      0 115.221.64.114:55980    101.230.149.37:11565    TIME_WAIT   -
tcp        0      0 115.221.64.114:58412    101.230.149.37:11565    ESTABLISHED 1107/clash
tcp        0      0 115.221.64.114:49496    101.230.149.37:11565    ESTABLISHED 1107/clash
tcp        0      0 115.221.64.114:57910    101.230.149.37:11565    TIME_WAIT   -
tcp        0      0 115.221.64.114:57908    101.230.149.37:11565    TIME_WAIT   -
tcp        0      0 115.221.64.114:58726    101.230.149.37:11565    ESTABLISHED 1107/clash
tcp        0      0 115.221.64.114:58192    101.230.149.37:11565    TIME_WAIT   -
tcp        0      0 115.221.64.114:56044    101.230.149.37:11565    TIME_WAIT   -
tcp        0      0 115.221.64.114:55642    101.230.149.37:11565    TIME_WAIT   -
tcp        0      0 115.221.64.114:54484    101.230.149.37:11565    TIME_WAIT   -
tcp        0      0 115.221.64.114:58508    101.230.149.37:11565    ESTABLISHED 1107/clash
tcp        0      0 115.221.64.114:57460    101.230.149.37:11565    TIME_WAIT   -
tcp        0      0 115.221.64.114:57316    101.230.149.37:11565    TIME_WAIT   -
tcp        0      0 115.221.64.114:55928    101.230.149.37:11565    TIME_WAIT   -
tcp        0      0 115.221.64.114:55400    101.230.149.37:11565    TIME_WAIT   -
tcp        0      0 115.221.64.114:56180    101.230.149.37:11565    TIME_WAIT   -
tcp        0      0 115.221.64.114:56110    101.230.149.37:11565    TIME_WAIT   -
tcp        0      0 115.221.64.114:55976    101.230.149.37:11565    TIME_WAIT   -
tcp        0      0 115.221.64.114:56226    101.230.149.37:11565    TIME_WAIT   -
tcp        0      0 115.221.64.114:56086    101.230.149.37:11565    TIME_WAIT   -
tcp        0      0 115.221.64.114:56220    101.230.149.37:11565    TIME_WAIT   -
tcp        0      0 115.221.64.114:55376    101.230.149.37:11565    TIME_WAIT   -
tcp        0      0 115.221.64.114:57406    101.230.149.37:11565    TIME_WAIT   -
tcp        0      0 115.221.64.114:57664    101.230.149.37:11565    ESTABLISHED 1107/clash
tcp        0      0 115.221.64.114:55998    101.230.149.37:11565    ESTABLISHED 1107/clash
tcp        0      0 115.221.64.114:56676    101.230.149.37:11565    TIME_WAIT   -
tcp        0      0 115.221.64.114:57472    101.230.149.37:11565    TIME_WAIT   -
tcp        0      0 115.221.64.114:56312    101.230.149.37:11565    TIME_WAIT   -
[... cut ...]
tcp        0      0 115.221.64.114:56556    101.230.149.37:11565    TIME_WAIT   -
tcp        0      0 192.168.2.1:33668       192.168.2.12:3389       FIN_WAIT2   -
tcp        0      0 115.221.64.114:58724    101.230.149.37:11565    ESTABLISHED 1107/clash
tcp        0      0 115.221.64.114:57352    101.230.149.37:11565    TIME_WAIT   -
tcp        0      0 115.221.64.114:57702    101.230.149.37:11565    TIME_WAIT   -
tcp        0      0 115.221.64.114:57702    101.230.149.37:11565
```
</pre>
</details>
不知道这个算不算是被扫描了，每次列出的ip还都不一定一样

hosence, 2022-02-03:
现在已经用2022.1.31正式版两天了，很好用。

Sorkai, 2022-02-06:
[image]
因为smartdns一点都不smart我就把smartdns关掉了，然后就发生了这个问题，一直也没怀疑这个已经关上的smartdns，今天我看那些udp连接的目标地址都是smartdns列表中的(其它dns程序并未使用那几个地址)，就把smartdns配置一下打开了，，，，，然后，，竟然好了，，，，，，现在的状况就是只要不开smartdns就会出现上面UDP连接异常的问题，但我真的不想不用那smartdns

Nxir, 2022-02-11:
> > 20220103无docker正式版 同样遇到掉速问题 重启后正常 服务只启动了openclash
> 
> 我把网络共享samba关掉后(系统->启动项->samba->禁用) 暂时看来情况有好转 之前几个小时就会掉速 现在半天了还没掉 具体原因不清楚 再观察观察
> 
> 另外 大多数时间我的在线用户数在35-40之间 我留意到 samba关闭之前活动连接数一直有两三千 关闭后就维持在一千多

这么些天过去了  发现还是掉速  只是掉的比较慢了

ps: 已经更新到2022.1.31

closed 2022-03-18 by DHDAXCW

Xlyaon, 2022-06-25:
> 目前刷了20220103的正式版，稳定运行5天了，没有掉速。20220101的也不会掉速。但是群里有人说他刷的20220101版本掉速。这到底是哪的问题我也没弄明白。

请问20220103的固件还有存档吗？不小心删掉了现在下不到了

hosence, 2022-06-27:
https://drive.google.com/drive/folders/1T0z3wrGzPzrHha0aXjw4jNhMdWAhhRs9?usp=sharing

Xlyaon, 2022-07-08:
谢谢分享