DHDAXCW/NanoPi-R4S-R4SE #168: R4S 内核出错后，网络正常，但无法进入后台。

happyghw, 2022-01-25:
Tue Jan 25 22:17:12 2022 kern.alert kernel: [79098.506536] Unable to handle kernel paging request at virtual address dfff0000e86c7070
Tue Jan 25 22:17:12 2022 kern.alert kernel: [79098.507240] Mem abort info:
Tue Jan 25 22:17:12 2022 kern.alert kernel: [79098.507488]   ESR = 0x96000004
Tue Jan 25 22:17:12 2022 kern.alert kernel: [79098.507762]   EC = 0x25: DABT (current EL), IL = 32 bits
Tue Jan 25 22:17:12 2022 kern.alert kernel: [79098.508226]   SET = 0, FnV = 0
Tue Jan 25 22:17:12 2022 kern.alert kernel: [79098.508494]   EA = 0, S1PTW = 0
Tue Jan 25 22:17:12 2022 kern.alert kernel: [79098.508768] Data abort info:
Tue Jan 25 22:17:12 2022 kern.alert kernel: [79098.509021]   ISV = 0, ISS = 0x00000004
Tue Jan 25 22:17:12 2022 kern.alert kernel: [79098.509680]   CM = 0, WnR = 0
Tue Jan 25 22:17:12 2022 kern.alert kernel: [79098.509951] [dfff0000e86c7070] address between user and kernel address ranges
Tue Jan 25 22:17:12 2022 kern.emerg kernel: [79098.510580] Internal error: Oops: 96000004 [#1] PREEMPT SMP
Tue Jan 25 22:17:12 2022 kern.warn kernel: [79098.511070] Modules linked in: shortcut_fe_cm xt_FULLCONENAT ath9k_htc ath9k_common rtw88_8822ce rtw88_8822c rtw88_8822be rtw88_8822b rtw88_8723de rtw88_8723d rtl8821ae rtl8192se rtl8192de rtl8192cu rtl8192ce rtl8192c_common rtl_usb rtl_pci rt2800usb rt2800lib rt2500usb pppoe ppp_async option mt76x0u mt76x0_common l2tp_ppp cdc_mbim carl9170 btcoexist ath9k_hw ath6kl_usb ath6kl_core ath zd1211rw wireguard usb8xxx usb_wwan sr9700 snd_usb_audio smsc95xx sierra_net rtw88_pci rtw88_core rtlwifi rtl8xxxu rtl88x2bu rtl8821cu rtl8812au rtl818x_pci rtl8188eu rtl8187 rt73usb rt2x00usb rt2x00lib rsi_usb rsi_91x rndis_host r8723bs qmi_wwan pptp pppox ppp_mppe ppp_generic plusb p54usb p54common mt76x2u mt76x2_common mt76x02_usb mt76x02_lib mt7663u mt7663_usb_sdio_common mt7615_common mt7603e mt7601u mt76_usb mt76_connac_lib mt76 mcs7830 mac80211 libertas libchacha20poly1305 libblake2s kalmia ipt_REJECT huawei_cdc_ncm dm9601 chacha_neon cfg80211 cdc_subset cdc_ncm cdc_ether cdc_eem ax88179_178a asix
Tue Jan 25 22:17:12 2022 kern.warn kernel: [79098.511181]  zstd xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_socket xt_recent xt_quota xt_policy xt_pkttype xt_owner xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_iprange xt_hl xt_helper xt_esp xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_cgroup xt_addrtype xt_TPROXY xt_TCPMSS xt_REDIRECT xt_NETMAP xt_MASQUERADE xt_LOG xt_HL xt_DSCP xt_CT xt_CLASSIFY usbserial usbnet usblp ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda ts_fsm ts_bm tcp_bbr snd_usbmidi_lib slhc sch_cake rtl8150 r8712u(C) r8169 r8152 poly1305_neon pegasus nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject_bridge nft_reject nft_redir nft_quota nft_objref nft_numgen nft_meta_bridge nft_log nft_limit nft_hash nft_fwd_netdev nft_dup_netdev nft_ct nft_counter nf_tproxy_ipv6 nf_tproxy_ipv4 nf_tables_set nf_tables nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip
Tue Jan 25 22:17:12 2022 kern.warn kernel: [79098.518833]  nf_nat_pptp nf_nat_irc nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_log_ipv4 nf_dup_netdev nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_pptp nf_conntrack_netlink nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast ts_kmp nf_conntrack_amanda nf_conncount macvlan lzo libcurve25519_generic libchacha libblake2s_generic lib80211_crypt_wep lib80211_crypt_tkip lib80211_crypt_ccmp lib80211 kaweth iptable_raw iptable_nat iptable_mangle iptable_filter ipt_ah ipt_ECN ipheth ip_tables hso crc_itu_t crc_ccitt compat cdc_wdm cdc_acm asn1_decoder fuse sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_tcindex cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred act_gact configs cryptodev xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip
Tue Jan 25 22:17:12 2022 kern.warn kernel: [79098.526453]  ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink net2280 ip6table_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 ip6t_NPT nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 nfsv4 nfsv3 nfsd nfs bonding ip_gre gre ifb dummy sit l2tp_netlink l2tp_core ipcomp6 xfrm6_tunnel esp6 ah6 xfrm4_tunnel ipcomp esp4 ah4 tunnel6 tunnel4 ip_tunnel rpcsec_gss_krb5 auth_rpcgss oid_registry veth tun snd_rawmidi snd_seq_device snd_pcm_oss snd_pcm snd_timer snd_mixer_oss snd_hwdep snd_compress snd soundcore xfrm_user xfrm_ipcomp af_key xfrm_algo vfat fat lockd sunrpc grace cifs dns_resolver dm_mirror dm_region_hash dm_log dm_crypt dm_mod dax raid456 async_raid6_recov async_pq async_xor async_memcpy async_tx raid10 raid1 raid0 md_mod nls_utf8 nls_cp437 eeprom_93cx6 vxlan udp_tunnel ip6_udp_tunnel shortcut_fe_ipv6 shortcut_fe crypto_user algif_skcipher algif_rng algif_hash algif_aead af_alg sha512_generic
Tue Jan 25 22:17:12 2022 kern.warn kernel: [79098.534049]  sha1_generic seqiv md5 md4 kpp ghash_generic gf128mul gcm echainiv ecb des_generic libdes deflate cts ctr cmac ccm authenc arc4 crypto_acompress uas dwc2 ohci_pci uhci_hcd fsl_mph_dr_of ehci_fsl gpio_button_hotplug exfat(C) btrfs xor zstd_decompress zstd_compress xor_neon raid6_pq lzo_decompress lzo_compress antfs udc_core cbc encrypted_keys trusted tpm
Tue Jan 25 22:17:12 2022 kern.warn kernel: [79098.544406] CPU: 0 PID: 18848 Comm: ubus Tainted: G         C        5.4.162 #0
Tue Jan 25 22:17:12 2022 kern.warn kernel: [79098.545044] Hardware name: FriendlyElec NanoPi R4S (DT)
Tue Jan 25 22:17:12 2022 kern.warn kernel: [79098.545504] pstate: 80000005 (Nzcv daif -PAN -UAO)
Tue Jan 25 22:17:12 2022 kern.warn kernel: [79098.545932] pc : vma_interval_tree_remove+0x170/0x2c8
Tue Jan 25 22:17:12 2022 kern.warn kernel: [79098.546377] lr : __vma_adjust+0x3c0/0x790
Tue Jan 25 22:17:12 2022 kern.warn kernel: [79098.546728] sp : ffff8000177a3af0
Tue Jan 25 22:17:12 2022 kern.warn kernel: [79098.547018] x29: ffff8000177a3af0 x28: 0000ffffb19e6000
Tue Jan 25 22:17:12 2022 kern.warn kernel: [79098.547482] x27: ffff0000aa87b000 x26: 0000000000000000
Tue Jan 25 22:17:12 2022 kern.warn kernel: [79098.547947] x25: ffff0000f0a655c8 x24: ffff0000673da300
Tue Jan 25 22:17:12 2022 kern.warn kernel: [79098.548411] x23: 0000000000000000 x22: 0000000000000000
Tue Jan 25 22:17:12 2022 kern.warn kernel: [79098.548875] x21: ffff0000ab2222e0 x20: ffff0000ab222338
Tue Jan 25 22:17:12 2022 kern.warn kernel: [79098.549339] x19: ffff0000f0a655c8 x18: 0000000000000000
Tue Jan 25 22:17:12 2022 kern.warn kernel: [79098.549803] x17: 0000000000000000 x16: 0000000000000000
Tue Jan 25 22:17:12 2022 kern.warn kernel: [79098.550266] x15: 0000000000000000 x14: ffff0000ab222358
Tue Jan 25 22:17:12 2022 kern.warn kernel: [79098.550730] x13: 0000000000000013 x12: 0000000000000000
Tue Jan 25 22:17:12 2022 kern.warn kernel: [79098.551194] x11: 0000000000000000 x10: ffff0000e85db4a8
Tue Jan 25 22:17:12 2022 kern.warn kernel: [79098.551658] x9 : 0000000000000075 x8 : 0000000000000000
Tue Jan 25 22:17:12 2022 kern.warn kernel: [79098.552122] x7 : 0000000000000012 x6 : ffff0000e6dc1e00
Tue Jan 25 22:17:12 2022 kern.warn kernel: [79098.552585] x5 : 0000ffff84325000 x4 : dfff0000e86c7058
Tue Jan 25 22:17:12 2022 kern.warn kernel: [79098.553050] x3 : ffff0000e9c8df18 x2 : ffffffffffffffff
Tue Jan 25 22:17:12 2022 kern.warn kernel: [79098.553514] x1 : 0000000000000011 x0 : 0000000000000000
Tue Jan 25 22:17:12 2022 kern.warn kernel: [79098.553978] Call trace:
Tue Jan 25 22:17:12 2022 kern.warn kernel: [79098.554197]  vma_interval_tree_remove+0x170/0x2c8
Tue Jan 25 22:17:12 2022 kern.warn kernel: [79098.554609]  __vma_adjust+0x3c0/0x790
Tue Jan 25 22:17:12 2022 kern.warn kernel: [79098.554930]  __split_vma+0x158/0x178
Tue Jan 25 22:17:12 2022 kern.warn kernel: [79098.555243]  __do_munmap+0xd4/0x4e8
Tue Jan 25 22:17:12 2022 kern.warn kernel: [79098.555550]  mmap_region+0x20c/0x630
Tue Jan 25 22:17:12 2022 kern.warn kernel: [79098.555864]  do_mmap+0x304/0x4b8
Tue Jan 25 22:17:12 2022 kern.warn kernel: [79098.556151]  vm_mmap_pgoff+0x7c/0x100
Tue Jan 25 22:17:12 2022 kern.warn kernel: [79098.556472]  ksys_mmap_pgoff+0x158/0x1f0
[... cut ...] 56819]  __arm64_sys_mmap+0x28/0x38
Tue Jan 25 22:17:12 2022 kern.warn kernel: [79098.557158]  el0_svc_common.constprop.2+0x7c/0x110
Tue Jan 25 22:17:12 2022 kern.warn kernel: [79098.557577]  el0_svc_handler+0x20/0x80
Tue Jan 25 22:17:12 2022 kern.warn kernel: [79098.557905]  el0_svc+0x8/0x200
Tue Jan 25 22:17:12 2022 kern.emerg kernel: [79098.558178] Code: d1000442 cb050021 8b413041 b4000084 (f9400c82)
Tue Jan 25 22:17:12 2022 kern.warn kernel: [79098.558712] ---[ end trace 2ed319349886a394 ]---

DHDAXCW, 2022-01-26:
刷最新固件

happyghw, 2022-01-26:
目前是 0111版，因内核崩溃问题一直困扰着我，自使用 R4S 以来一直存在，而且是不定时，然后刚才 21：16 又崩溃了。

closed 2022-03-18 by DHDAXCW