DHDAXCW/NanoPi-R4S-R4SE #182: 8812bu无线网卡不稳定

mobier, 2022-03-03:
固件版本	DHDAXCW @ FusionWrt R22.2.2 (2022-02-19) / LuCI Master (git-22.046.49039-d9baa31)
内核版本	5.4.175
WIFI网卡芯片RTL8812bu
问题1：连接上热点后speedtest测速1~2次后无线直接崩了界面提示“无线未开启或未关联”，经常上着网不测速在一定时间后也会崩怀疑也是这个原因
问题2：无线无法以client角色连接其他wifi，界面提示如问题1
使用官方固件：rk3399-sd-friendlywrt-5.15-20220125.img.zip 则没有以上问题
内核日志：
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] Linux version 5.4.175 (runner@fv-az139-451) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r4213-2cef640d5)) #0 SMP PREEMPT Sat Feb 19 01:59:00 2022
[    0.000000] Machine model: FriendlyElec NanoPi R4S
[    0.000000] earlycon: uart8250 at MMIO32 0x00000000ff1a0000 (options '')
[    0.000000] printk: bootconsole [uart8250] enabled
[    0.000000] cma: Reserved 8 MiB at 0x00000000f7800000
[    0.000000] On node 0 totalpages: 1015296
[    0.000000]   DMA32 zone: 15864 pages used for memmap
[    0.000000]   DMA32 zone: 0 pages reserved
[    0.000000]   DMA32 zone: 1015296 pages, LIFO batch:63
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.0 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] psci: SMC Calling Convention v1.0
[    0.000000] percpu: Embedded 20 pages/cpu s44952 r8192 d28776 u81920
[    0.000000] pcpu-alloc: s44952 r8192 d28776 u81920 alloc=20*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 [0] 4 [0] 5 
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: detected: ARM erratum 845719
[    0.000000] CPU features: detected: GIC system register CPU interface
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 999432
[    0.000000] Kernel command line: console=ttyS2,1500000 earlycon=uart8250,mmio32,0xff1a0000 root=PARTUUID=6391314e-02 rw rootwait
[    0.000000] Dentry cache hash table entries: 524288 (order: 10, 4194304 bytes, linear)
[    0.000000] Inode-cache hash table entries: 262144 (order: 9, 2097152 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 3959200K/4061184K available (7742K kernel code, 576K rwdata, 2412K rodata, 1728K init, 729K bss, 93792K reserved, 8192K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=6, Nodes=1
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu: 	RCU event tracing is enabled.
[    0.000000] rcu: 	RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=6.
[    0.000000] 	Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=6
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] GICv3: GIC: Using split EOI/Deactivate mode
[    0.000000] GICv3: 256 SPIs implemented
[    0.000000] GICv3: 0 Extended SPIs implemented
[    0.000000] GICv3: Distributor has no Range Selector support
[    0.000000] GICv3: 16 PPIs implemented
[    0.000000] GICv3: no VLPI support, no direct LPI support
[    0.000000] GICv3: CPU0: found redistributor 0 region 0:0x00000000fef00000
[    0.000000] ITS [mem 0xfee20000-0xfee3ffff]
[    0.000000] ITS@0x00000000fee20000: allocated 65536 Devices @f2080000 (flat, esz 8, psz 64K, shr 0)
[    0.000000] ITS: using cache flushing for cmd queue
[    0.000000] GICv3: using LPI property table @0x00000000f2040000
[    0.000000] GIC: using cache flushing for LPI property table
[    0.000000] GICv3: CPU0: using allocated LPI pending table @0x00000000f2050000
[    0.000000] GICv3: GIC: PPI partition interrupt-partition-0[0] { /cpus/cpu@0[0] /cpus/cpu@1[1] /cpus/cpu@2[2] /cpus/cpu@3[3] }
[    0.000000] GICv3: GIC: PPI partition interrupt-partition-1[1] { /cpus/cpu@100[4] /cpus/cpu@101[5] }
[    0.000000] random: get_random_bytes called from start_kernel+0x29c/0x39c with crng_init=0
[    0.000000] arch_timer: cp15 timer(s) running at 24.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x588fe9dc0, max_idle_ns: 440795202592 ns
[    0.000018] sched_clock: 56 bits at 24MHz, resolution 41ns, wraps every 4398046511097ns
[    0.003864] Console: colour dummy device 80x25
[    0.004417] Calibrating delay loop (skipped), value calculated using timer frequency.. 48.00 BogoMIPS (lpj=240000)
[    0.005462] pid_max: default: 32768 minimum: 301
[    0.006512] Mount-cache hash table entries: 8192 (order: 4, 65536 bytes, linear)
[    0.007394] Mountpoint-cache hash table entries: 8192 (order: 4, 65536 bytes, linear)
[    0.013327] ASID allocator initialised with 32768 entries
[    0.014203] rcu: Hierarchical SRCU implementation.
[    0.015453] Platform MSI: interrupt-controller@fee20000 domain created
[    0.017021] PCI/MSI: /interrupt-controller@fee00000/interrupt-controller@fee20000 domain created
[... 463 lines cut ...]
[   16.799441] EXT4-fs (mmcblk0p1): warning: mounting unchecked fs, running e2fsck is recommended
[   16.807388] EXT4-fs (mmcblk0p1): mounted filesystem without journal. Opts: 
[   17.415565] eth1: 0xffff800011217000, 6a:27:19:a5:40:87, IRQ 232
[   17.474637] br-lan: port 1(eth1) entered blocking state
[   17.475099] br-lan: port 1(eth1) entered disabled state
[   17.475680] device eth1 entered promiscuous mode
[   17.479748] phy_rtl8211e_led_fixup in
[   17.481564] rk_gmac-dwmac fe300000.ethernet eth0: PHY [stmmac-0:01] driver [RTL8211E Gigabit Ethernet]
[   19.049745] rk_gmac-dwmac fe300000.ethernet eth0: No Safety Features support found
[   19.050416] rk_gmac-dwmac fe300000.ethernet eth0: PTP not supported by HW
[   19.051011] rk_gmac-dwmac fe300000.ethernet eth0: configuring for phy/rgmii link mode
[   20.575774] r8168: eth1: link up
[   20.576183] br-lan: port 1(eth1) entered blocking state
[   20.576641] br-lan: port 1(eth1) entered forwarding state
[   20.577219] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[   21.459351] start_addr=(0x20000), end_addr=(0x40000), buffer_size=(0x20000), smp_number_max=(16384)
[   21.579668] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   38.191710] vbus_typec: disabling
[   41.108569] NFSD: Using /var/lib/nfs/v4recovery as the NFSv4 state recovery directory
[   41.109317] NFSD: Using legacy client tracking operations.
[   41.109812] NFSD: starting 10-second grace period (net f00000a0)
[   95.213951] xhci-hcd xhci-hcd.0.auto: xHCI host not responding to stop endpoint command.
[   95.228269] xhci-hcd xhci-hcd.0.auto: xHCI host controller not responding, assume dead
[   95.229138] xhci-hcd xhci-hcd.0.auto: HC died; cleaning up
[   95.229828] usb 6-1: USB disconnect, device number 2

DHDAXCW, 2022-03-17:
已修复，等坐3.17

closed 2022-03-17 by DHDAXCW

mobier, 2022-03-17:
固件版本 | DHDAXCW @ FusionWrt R22.3.13 (2022-03-17) / LuCI Master (git-22.068.45502-a50e601)
内核版本 | 5.4.185
已升级到3.17版本依旧是相同问题

speadk888, 2022-05-03:
请问是哪个版本呀，为啥我的同款的无线网卡识别不出来