DHDAXCW/NanoPi-R4S-R4SE #200: 通过TTYD终端在线升级，无最新版本，只能找到多天前的版本？

reallwp, 2022-04-02:


Fetiz, 2022-04-02:
刚刚测试更新脚本，直接提示无法找到固件
[image]

reallwp, 2022-04-02:
> 刚刚测试更新脚本，直接提示无法找到固件 [image]

你发完我刚试，我可以找到固件，但还是不能找到当天的版本
[image]
[image]

Fetiz, 2022-04-02:
> > 刚刚测试更新脚本，直接提示无法找到固件 [image]
> 
> 
> 
> 你发完我刚试，我可以找到固件，但还是不能找到当天的版本
> 
> [image]
> 
> [image]
> 
> 
> 
> 

去看了眼release 发现这几天的release 都没有生成固件 全是源码。估计是这个原因

DHDAXCW, 2022-04-02:
今天版本编译有点问题，全改明天出包

reallwp, 2022-04-02:
> 今天版本编译有点问题，全改明天出包

谢谢大佬

ericghost, 2022-04-03:
感觉还是3.18这个用着最稳定,目前对我来说

reallwp, 2022-04-03:
> 感觉还是3.18这个用着最稳定,目前对我来说

大佬把3.18后的都删了🥶

ericghost, 2022-04-03:
> 
哈哈,刚发现,后面几天的固件我都刷了, 插件不全,而且还有点bug,估计就是编译的问题吧, 也就18号这个还行反正用着没啥问题

closed 2022-04-06 by reallwp