DHDAXCW/NanoPi-R4S-R4SE #203: r4s 5.15内核测试，有问题反馈到这里

DHDAXCW, 2022-04-05:


DHDAXCW, 2022-04-05:
r4s dn2 内核升级到5.15 ，支持无线，尽量反馈问题啊，每天晚上修复，r2s dn1话不多说了，直接上链接https://drive.google.com/drive/folders/1WnHAMoNNinEGOJ9qcbdoImP2gJWPLqmh?usp=sharing    睡觉去了。。。。

ericghost, 2022-04-06:
刚装好,暂时没发现什么大问题

就是配置openclash的时候 有时候需要重启路由器才能正常访问外网

DHDAXCW, 2022-04-06:
> 刚装好，暂时没发现什么大问题
> 
> 配置openclash的时候就是需要外网路由器才能正常访问
> 
> 刚adgh内核更新后仍然显示内核，无法发现错误error!bin path is a dir 原因找不到文件路径错误，默认执行的路径是/usr/bin/AdGuardHome/AdGuardHome，后/usr/bin/AdGuardHome /AdGuardHome,然后就正常了

ok

ericghost, 2022-04-06:
> 

我自己装了smartdns ipk  暂时没有什么稳定性的问题,用了一下午了,等过几天再回复

DHDAXCW, 2022-04-06:
> > 
> 
> 我自己装了smartdns ipk 暂时没有什么稳定性的问题,用了一下午了,等过几天再回复

今晚加回来smartdns

quango-web, 2022-04-07:
R4s-电信300M上行30。HW插件vm+aes加密，使用speed同一节点测多次，
4.06版本下行最高206，上行33，另HW插件点击更新国内IP数据库没反应，4.02可以
4.02版本下行维持360-380，下行达到峰值非常快，上行基本不超过11

DHDAXCW, 2022-04-08:
> R4s-电信300M上行30。HW插件vm+aes加密，使用speed同一节点测多次， 4.06版本下行最高206，上行33，另HW插件点击更新国内IP数据库没反应，4.02可以 4.02版本下行维持360-380，下行达到峰值非常快，上行基本不超过11

已找到了，下午GitHub发

ericghost, 2022-04-09:
ssrp 还有 HW 好像绕过大陆ip有问题,会变成全局代理, 设置GFW模式后正常

ericghost, 2022-04-09:
又出现看国内视频的时候,出现地域限制, 用的是ssrplus+绕过中国大陆, 用HW就没有这个问题,但是HW不能更新国内IP段,点击无反应, 最早你网盘里的那个5.15内核的固件还在吗? 我想再下载一下

fzudust, 2022-04-10:
ssrplus是不是分流有问题，绕过中国大陆是全局代理

Zypeong, 2022-04-10:
lan端设备无法联网，可以ssh进路由器，网页无法进入管理页面。

zmnsQ, 2022-04-12:
新内核还会出Docker版吗

mimosa, 2022-04-12:
5.15内核的 `kmod-hwmon-pwmfan` 哪里可以下载到？

DHDAXCW, 2022-04-12:
这个在x86上可加的，你直接usb就好了

mimosa, 2022-04-12:
> 这个在x86上可加的，你直接usb就好了

买了大佬 readme里推荐的 风扇链接 [点击进入](https://s.click.taobao.com/t?e=m%3D2%26s%3Dd8Ack0Lbx8McQipKwQzePOeEDrYVVa64LKpWJ%2Bin0XJRAdhuF14FMXpyNmcFd6mT8sviUM61dt2T0mcOGN1M6FAj1gqltKaEfKzCcEr0EW0YuhTK3FPxiHMT7yc3NZrQKSOkJV8harV3phaPbavinqGCwVfdcN0wcSpj5qSCmbA%3D) 建议用这种风扇

pwm 温控风扇 给 R4S用。

DHDAXCW, 2022-04-13:
> > 这个在x86上可加的，你直接usb就好了
> 
> 买了大佬 readme里推荐的 风扇链接 [点击进入](https://s.click.taobao.com/t?e=m%3D2%26s%3Dd8Ack0Lbx8McQipKwQzePOeEDrYVVa64LKpWJ%2Bin0XJRAdhuF14FMXpyNmcFd6mT8sviUM61dt2T0mcOGN1M6FAj1gqltKaEfKzCcEr0EW0YuhTK3FPxiHMT7yc3NZrQKSOkJV8harV3phaPbavinqGCwVfdcN0wcSpj5qSCmbA%3D) 建议用这种风扇
> 
> pwm 温控风扇 给 R4S用, 需要用到 `fa-rk3399-pwmfan`。
> 
> ```
> Tue Apr 12 23:13:42 2022 user.notice fa-rk3399-pwmfan[9679]: temp: 39375, duty: 5000, 20%
> Tue Apr 12 23:14:42 2022 user.notice fa-rk3399-pwmfan[9679]: temp: 40555, duty: 5000, 20%
> Tue Apr 12 23:15:42 2022 user.notice fa-rk3399-pwmfan[9679]: temp: 39375, duty: 5000, 20%
> Tue Apr 12 23:16:42 2022 user.notice fa-rk3399-pwmfan[9679]: temp: 39375, duty: 5000, 20%
> ```
> 
> ```shell
> #!/bin/bash
> 
> # initialize the fan speed value
> echo 0 > /sys/class/pwm/pwmchip1/export;
> sleep 1
> echo 0 > /sys/class/pwm/pwmchip1/pwm0/enable;
> echo 50000 > /sys/class/pwm/pwmchip1/pwm0/period;
> echo 1 > /sys/class/pwm/pwmchip1/pwm0/enable;
> 
> # 5 gears fan speed
> declare -a Percents=(100 80 60 40 20)
> # cpu temperature series
> declare -a CpuTemps=(56000 53000 48000 42000 36000)
> # fan speed series
> declare -a PwmDutyCycles=(1000 2000 3000 4000 5000)
> 
> # default fan is stopped
> DefaultDuty=49990
> DefaultPercents=0
> 
> while true; do
>     # get cpu temperature
>     temp=$(cat /sys/class/thermal/thermal_zone0/temp)
>     # initialize default value
>     INDEX=0
>     FOUNDTEMP=0
>     DUTY=$DefaultDuty
>     PERCENT=$DefaultPercents
> 
>     # pairing of temperature and speed
>     for i in 0 1 2 3 4; do
>         if [ $temp -gt ${CpuTemps[$i]} ]; then
>             INDEX=$i
>             FOUNDTEMP=1
>             break
>         fi
>     done
> 
>     # turn on fan rotation
>     if [ ${FOUNDTEMP} == 1 ]; then
>         DUTY=${PwmDutyCycles[$i]}
>         PERCENT=${Percents[$i]}
>     fi
> 
>     # override value for fan speed
>     echo $DUTY > /sys/class/pwm/pwmchip1/pwm0/duty_cycle;
> 
>     logger -t "fa-rk3399-pwmfan[$$]" "temp: $temp, duty: $DUTY, ${PERCENT}%"
> 
>     # heartbeat
>     sleep 60s;
> done
> ```

https://github.com/DHDAXCW/NanoPi-R4S/commit/5eccf35e575ef5e9edd2c6f1e79237d7b5232ce0

xxinyyy2014, 2022-04-17:
请问可否单独编译一个把ikoolproxy换成adbyby plus的固件，前者个人使用完全没有效果

DHDAXCW, 2022-04-18:
> 请问可否单独编译一个把ikoolproxy换成adbyby plus的固件，前者个人使用完全没有效果

下个版本

MW8023, 2022-04-19:
> 每天早上起床都发现R4S自动关机了，冰凉冰凉的🥶

关机是因为每天凌晨4点的时候会跑分，5.15的内核跑分带来的功率更高了我原本的5V2A就不够用了，换了高点的电流就行了，你如果要换的话 最起码5V3A起步

bububa, 2022-04-19:
> > 每天早上起床都发现R4S自动关机了，冰凉冰凉的🥶
> 
> 关机是因为每天凌晨4点的时候会跑分，5.15的内核跑分带来的功率更高了我原本的5V2A就不够用了，换了高点的电流就行了，你如果要换的话 最起码5V3A起步

恩，发现了，我把crontab里跑分关了

iScott93, 2022-04-20:
插着usb wifi网卡 （cf-811ac) 无线设置会死机，插着网卡 断电重启系统灯会一直闪进不来系统，要拔掉重启才可以

xspeed1989, 2022-04-26:
非DOCKER版 UDP转发无效
C:\Users\chang> nslookup  google.com 8.8.4.4
DNS request timed out.
    timeout was 2 seconds.
Server:  UnKnown
Address:  8.8.4.4

DNS request timed out.
    timeout was 2 seconds.
DNS request timed out.

DHDAXCW, 2022-04-27:
> 非DOCKER版 UDP转发无效 C:\Users\chang> nslookup google.com 8.8.4.4 DNS request timed out. timeout was 2 seconds. Server: UnKnown Address: 8.8.4.4
> 
> DNS request timed out. timeout was 2 seconds. DNS request timed out.

DHDAXCW @ FusionWrt R22.4.1 (2022-04-25) / LuCI Master (git-22.112.67846-7685242)  
C:\Users\caw08>nslookup google.com 8.8.4.4
服务器:  dns.google
Address:  8.8.4.4

非权威应答:
名称:    google.com
Address:  142.250.66.78

xspeed1989, 2022-04-27:
> > 非DOCKER版 UDP转发无效 C:\Users\chang> nslookup google.com 8.8.4.4 DNS request timed out. timeout was 2 seconds. Server: UnKnown Address: 8.8.4.4
> > DNS request timed out. timeout was 2 seconds. DNS request timed out.
> 
> **DHDAXCW** @ FusionWrt R22.4.1 (**2022-04-25**) / LuCI Master (git-22.112.67846-7685242) 内核版本 | **5.15.33**
> 
> C:\Users\caw08>nslookup google.com 8.8.4.4 服务器: dns.google Address: 8.8.4.4
> 
> 非权威应答: 名称: google.com Address: 142.250.66.78

奇怪，我这边用的是[shadowoscks](https://github.com/shadowsocks/openwrt-shadowsocks),
[image]

如果是nslookup google.com 192.168.2.1#5300 可以，
nslookup google.com 8.8.4.4就不行，
确认ss所依赖的kmod-ipt-tproxy也是安装好了的。

dfldylan, 2022-05-05:
> 插着usb wifi网卡 （cf-811ac) 无线设置会死机，插着网卡 断电重启系统灯会一直闪进不来系统，要拔掉重启才可以

有同样的问题

violarulan, 2022-05-15:
`/shin/ifup wan` 会导致 br-lan 也一并重启，随后可以拨号获取ip地址但是无法上网（路由器和LAN下设备均无法上网）

l4326769, 2022-05-16:
> 像绕过大陆ip有问题,会变成全局代理, 设

我这边5.09版本也遇到这个问题了

magiclan, 2022-05-29:
[2022.05.08 x86-64-Linux-5.15-稳定精简 不支持i225 网卡，能否发布个支持的版本

violarulan, 2022-06-11:
> `/shin/ifup wan` 会导致 br-lan 也一并重启，随后可以拨号获取ip地址但是无法上网（路由器和LAN下设备均无法上网）
> 
> 使用的是乞丐版（因为只有它支持俺的AX88179）
> 
> ~update：把 upnp 关了就正常了~

经多次排查，问题在我的设备是阉割 EEPROM 的版本（非所谓企业版），两个物理网口的 mac 是一样的，手动在 op 里覆盖指定一个既解决问题

closed 2022-07-19 by DHDAXCW