DHDAXCW/NanoPi-R4S-R4SE #210: Docker not working

sharicov, 2022-04-14:
13.04.22 Build with docker: https://github.com/DHDAXCW/NanoPi-R4S/releases/download/2022.04.13-Lean1/openwrt-rockchip-armv8-friendlyarm_nanopi-r4s-ext4-sysupgrade.img.gz

Docker not working properly, when trying to run portainer getting following error message:

```
docker run -d -p 8000:8000 -p 9000:9000 -p 9443:9443 --name=portainer --restart=always    -v /var/run/docker.sock:/var/run/docker.sock   -v portainer_data:/data portainer/portainer-ce:latest

d59839679a8d34121ae3f448f7bc9b34301aa64b7b60175be14b4bbd7b157f6f

docker: Error response from daemon: failed to create endpoint portainer on network bridge: adding interface veth93c424e to bridge docker0 failed: could not find bridge docker0: route ip+net: no such network interface.
```

sharicov, 2022-04-14:
Fixed with enabling docker network interface on boot:

luci/admin/network/network/docker

And restarting docker

luci/admin/docker/overview/restart

Please fix in the build itself

MW8023, 2022-04-15:
Try changing the docker network type to host

closed 2022-07-19 by DHDAXCW