DHDAXCW/NanoPi-R4S-R4SE #211: R4S刷了4.13日期的稳定版本  断电问题

MW8023, 2022-04-15:
大佬你好，刷新版本之前也是用您的固件（精简版本）持续开机了137天这样，期间都没有断连过也更没断电过。然后出于对新版的好奇刷了一个4月13日的稳定版本，刷了过后用得挺好的也没出什么异常情况，可是今天早上起床的时候竟然发现R4S断电了，指示灯都不亮了，不知道这个问题是什么导致的，我该怎么排查呢？麻烦指教了

PS：还是往常的刷机方法，下载openwrt-rockchip-armv8-friendlyarm_nanopi-r4s-ext4-sysupgrade.img.gz通过Rufus软件烧录进三星EVO 64GB规格的TF卡，是没有保存之前的配置的，避开配置冲突的问题。

gao4306629, 2022-04-15:
是电源问题，在跑分时cpu全速后整体电源负载升高，然后你的电源功率不够限流保护了，然后就断电了。 用树莓派的3A很好很稳定。

MW8023, 2022-04-19:
> 是电源问题，在跑分时cpu全速后整体电源负载升高，然后你的电源功率不够限流保护了，然后就断电了。 用树莓派的3A很好很稳定。

谢谢  我现在换成了5V5A的电源  不仅在跑coremark的时候没有断电 还外挂了一个移动硬盘 也照常运行

cccbs124, 2022-04-20:
> 是电源，全速后全电源时长问题，然后你的电源功率限制流保护了，然后一直在跑分。

大佬，我也是遇到了这位朋友的问题，白天无论如何使用都是不断电的，反倒是早上起来发现r4s竟然断电了，那按理说应该是晚上没人上网，它的供电应该更不是问题。我用的是小米手机的5v3a电源。刷的最新版200多mb的ext文件系统的固件。不瞒您说，我现在用了一个别的开发者编译的固件，就没出现断电情况。但是功能上还是觉得您这个好用…QAQ

closed 2022-07-19 by DHDAXCW