DHDAXCW/NanoPi-R4S-R4SE #231: 超频版的稳定性咋样？

solaris111, 2022-05-30:
请教下作者和各位玩友，超频版的稳定性如何？有没有长期重载出现死机的情况？我是想借鉴下这个超频的opp参数用在ubuntu上。@[DHDAXCW](https://github.com/DHDAXCW)
原厂给出的VDD参数是不是太保守了？
[image]

DHDAXCW, 2022-05-31:
这个芯片手册推荐供电是1.25v
[image]

上游直接给大核心用1.32v供电了。。。
[image]

之前改过1.27v，就死机了。可能是体质问题吧
反正用1.32v没什么大不了

solaris111, 2022-05-31:
我看到了，就是感觉电压超的略高。请教下您，有没有做过压力测试，做了多久？cpu温度能到多少？

Robird, 2022-07-14:
别超，这两天江苏气温40度，几乎天天死机。我折腾改参数呢。

closed 2022-07-19 by DHDAXCW