DHDAXCW/NanoPi-R4S-R4SE #246: 您所有的新固件都无法启动

Wangshengli05, 2022-07-17:
您所有的新固件都无法启动..您自己测试了吗？他们都未能启动。只有 7 月 4 日的固件还可以。 7 月 4 日之后，所有固件都无法启动

DHDAXCW, 2022-07-17:
是7.17吗?

DHDAXCW, 2022-07-17:
都能启动呢、、、
[image]
[image]

Wangshengli05, 2022-07-17:
7 月 13 日、14 日、16 日、17 日。 所有 R4S 启动失败。 你可以试试看。 只有7月4日是可以的
王胜利

DHDAXCW, 2022-07-17:
也试了 14 16能启动

Wangshengli05, 2022-07-17:
试试看。 都失败了。  7月17日也失败了。

Wangshengli05, 2022-07-17:
删除 7 月 17 日。 我尝试了所有，但都失败了

DHDAXCW, 2022-07-17:
换卡了？

Fvpder, 2022-07-17:
试了一下7.17能正常启动，更换内存卡试试，长期使用建议更换mlc颗粒的内存卡

Wangshengli05, 2022-07-17:
我现在不会买新的TF卡。  MLC 卡很慢。  Slc 单层卡速度很快。 我有很多卡我无法再次添加新卡

closed 2022-07-17 by DHDAXCW

Fvpder, 2022-07-17:
The translation is terrible, that I cannot unserstand you. I was wondering if you are using the r4se version, which is developed for a new type with emmc flash inside.

Wangshengli05, 2022-07-17:
I use R4S.  I don't use r4se

Wangshengli05, 2022-10-11:
换什么卡？ 我有 4 张抽认卡，我都炸了，但没有用。 固件有问题