DHDAXCW/NanoPi-R4S-R4SE #265: 请帮忙看下 R4s间歇断网问题

MethodJiao, 2022-09-06:
我的r4s刷了您的固件，但是只要打开passwall就会有时候间歇断网

[image]

（我写了个程序定位了11点40分和12点19分断网，日志截图是我在12点19分截取的）

内核日志和系统日志附上如下

[image]
[image]

MethodJiao, 2022-09-06:
不知道是否是passwall插件和系统产生了什么样的反应，我也有试过openclash，情况依旧

MethodJiao, 2022-09-06:
[image]
系统版本供大神定位

DHDAXCW, 2022-09-07:
刷最新吧

DHDAXCW, 2022-09-07:
如果你一直跑下载，那他就炸了，。

MethodJiao, 2022-09-07:
> 如果你一直跑下载，那他就炸了，。

刷了最新目前还没问题，很神奇

closed 2022-09-17 by MethodJiao