DHDAXCW/NanoPi-R4S-R4SE #281: bootstrap主题

xhat, 2022-10-11:
系统默认删除了所有主题，能理解。但目前通过软件包安装的bootstrap主题有显示内容缺失。
bootstrap主题用了很多年，不再习惯其他主题。希望能让默认的bootstrap主题回归。

zonyitoo, 2022-10-11:
> 通过软件包安装的bootstrap主题

是指用opkg装的吗？那是软件源上面的版本问题？

xhat, 2022-10-11:
是opkg默认源，换官方源一样。
[image]

DHDAXCW, 2022-10-11:
果然，下版本加主题~

DHDAXCW, 2022-10-11:
> 是opkg默认源，换官方源一样。 [image]

你这出现缺失是使用了21.04版，我们这里只能用18.06版的主题

xhat, 2022-10-11:
> 你这出现缺失是使用了21.04版，我们这里只能用18.06版的主题

18.06一样缺失。我只是尝试了最新的21.04是一样的结果。
希望增加默认主题，有些老用户还是怀旧的哈