DHDAXCW/NanoPi-R4S-R4SE #285: 10.19全插件版本helloworld依然存在问题

quango-web, 2022-10-20:
贴上部分系统日志供大佬过目：
Thu Oct 20 11:54:42 2022 daemon.crit dnsmasq[1]: cannot access directory /tmp/dnsmasq.ssr: No such file or directory
Thu Oct 20 11:54:42 2022 daemon.crit dnsmasq[1]: FAILED to start up
Thu Oct 20 11:54:47 2022 daemon.crit dnsmasq[1]: cannot access directory /tmp/dnsmasq.ssr: No such file or directory
Thu Oct 20 11:54:47 2022 daemon.crit dnsmasq[1]: FAILED to start up
Thu Oct 20 11:54:47 2022 daemon.info procd: Instance dnsmasq::cfg01411c s in a crash loop 6 crashes, 0 seconds since last crash
Thu Oct 20 11:55:01 2022 cron.info crond[25763]: crond (busybox 1.35.0) started, log level 5
Thu Oct 20 11:55:07 2022 daemon.crit dnsmasq[1]: cannot access directory /tmp/dnsmasq.ssr: No such file or directory
Thu Oct 20 11:55:07 2022 daemon.crit dnsmasq[1]: FAILED to start up
Thu Oct 20 11:55:12 2022 daemon.crit dnsmasq[1]: cannot access directory /tmp/dnsmasq.ssr: No such file or directory
Thu Oct 20 11:55:12 2022 daemon.crit dnsmasq[1]: FAILED to start up
Thu Oct 20 11:55:17 2022 daemon.crit dnsmasq[1]: cannot access directory /tmp/dnsmasq.ssr: No such file or directory
Thu Oct 20 11:55:17 2022 daemon.crit dnsmasq[1]: FAILED to start up
Thu Oct 20 11:55:22 2022 daemon.crit dnsmasq[1]: cannot access directory /tmp/dnsmasq.ssr: No such file or directory
Thu Oct 20 11:55:22 2022 daemon.crit dnsmasq[1]: FAILED to start up
Thu Oct 20 11:55:27 2022 daemon.crit dnsmasq[1]: cannot access directory /tmp/dnsmasq.ssr: No such file or directory
Thu Oct 20 11:55:27 2022 daemon.crit dnsmasq[1]: FAILED to start up
Thu Oct 20 11:55:32 2022 daemon.crit dnsmasq[1]: cannot access directory /tmp/dnsmasq.ssr: No such file or directory
Thu Oct 20 11:55:32 2022 daemon.crit dnsmasq[1]: FAILED to start up
Thu Oct 20 11:55:32 2022 daemon.info procd: Instance dnsmasq::cfg01411c s in a crash loop 6 crashes, 0 seconds since last crash

DHDAXCW, 2022-10-20:
https://github.com/DHDAXCW/NanoPi-R4S-R4SE/releases/tag/2022.10.20-Lean3

quango-web, 2022-10-20:
10.20版本的HW依旧不行，故障表现反映在日志上条目一样。这次测试了SR+则没有问题。
我是外行，只能给大佬反映有限的直观表述，请包涵。

quango-web, 2022-10-21:
10.21全版本的HW不通，日志上一样:
Fri Oct 21 14:35:40 2022 cron.info crond[13942]: time disparity of 5130579 minutes detected
Fri Oct 21 14:36:03 2022 cron.info crond[23661]: crond (busybox 1.35.0) started, log level 5
Fri Oct 21 14:36:23 2022 cron.info crond[24318]: crond (busybox 1.35.0) started, log level 5
Fri Oct 21 14:36:31 2022 daemon.crit dnsmasq[1]: cannot access directory /tmp/dnsmasq.ssr: No such file or directory
Fri Oct 21 14:36:31 2022 daemon.crit dnsmasq[1]: FAILED to start up
Fri Oct 21 14:36:36 2022 daemon.crit dnsmasq[1]: cannot access directory /tmp/dnsmasq.ssr: No such file or directory
Fri Oct 21 14:36:36 2022 daemon.crit dnsmasq[1]: FAILED to start up
Fri Oct 21 14:36:41 2022 daemon.crit dnsmasq[1]: cannot access directory /tmp/dnsmasq.ssr: No such file or directory
Fri Oct 21 14:36:41 2022 daemon.crit dnsmasq[1]: FAILED to start up
Fri Oct 21 14:36:46 2022 daemon.crit dnsmasq[1]: cannot access directory /tmp/dnsmasq.ssr: No such file or directory
Fri Oct 21 14:36:46 2022 daemon.crit dnsmasq[1]: FAILED to start up
Fri Oct 21 14:36:51 2022 daemon.crit dnsmasq[1]: cannot access directory /tmp/dnsmasq.ssr: No such file or directory
Fri Oct 21 14:36:51 2022 daemon.crit dnsmasq[1]: FAILED to start up

quango-web, 2022-10-21:
10.21全固件除了改本地IP拨号上网外，其他都是保持出厂默认设置。
将HW的DNS默认8.8.4.4解析改为1.1.1.1后保存启动后也不行。
日志：
Fri Oct 21 14:36:56 2022 daemon.info procd: Instance dnsmasq::cfg01411c s in a crash loop 6 crashes, 0 seconds since last crash
Fri Oct 21 14:38:34 2022 cron.info crond[25916]: crond (busybox 1.35.0) started, log level 5
Fri Oct 21 14:38:40 2022 daemon.crit dnsmasq[1]: cannot access directory /tmp/dnsmasq.ssr: No such file or directory
Fri Oct 21 14:38:40 2022 daemon.crit dnsmasq[1]: FAILED to start up
Fri Oct 21 14:38:45 2022 daemon.crit dnsmasq[1]: cannot access directory /tmp/dnsmasq.ssr: No such file or directory
Fri Oct 21 14:38:45 2022 daemon.crit dnsmasq[1]: FAILED to start up
Fri Oct 21 14:38:51 2022 daemon.crit dnsmasq[1]: cannot access directory /tmp/dnsmasq.ssr: No such file or directory
Fri Oct 21 14:38:51 2022 daemon.crit dnsmasq[1]: FAILED to start up
Fri Oct 21 14:38:56 2022 daemon.crit dnsmasq[1]: cannot access directory /tmp/dnsmasq.ssr: No such file or directory
Fri Oct 21 14:38:56 2022 daemon.crit dnsmasq[1]: FAILED to start up

DHDAXCW, 2022-10-21:
你是开了两个？看起来是ssrplus才会去设/tmp/dnsmasq.ssr，把它给设到了dnsmasq的配置里
不会又开了那个ssrplus的插件又开helloworld吧？

quango-web, 2022-10-21:
没有开两个，这个常识我还是有的。前几个版本及今天新刷10.21全插件版本，都先直接设定拨号跟HW，其他插件一概保持原固件默认。
我已经退回9.19全插件版本，这个版本使用HW就非常正常稳定！且加装了smartdns配合也没问题。

quango-web, 2022-10-22:
10.22全固件，HW依旧。SR+则ok，
[image]
[image]
难道是HW版本问题？
Sat Oct 22 21:29:56 2022 cron.info crond[23992]: crond (busybox 1.35.0) started, log level 5
Sat Oct 22 21:30:45 2022 cron.info crond[24851]: crond (busybox 1.35.0) started, log level 5
Sat Oct 22 21:30:53 2022 daemon.crit dnsmasq[1]: cannot access directory /tmp/dnsmasq.ssr: No such file or directory
Sat Oct 22 21:30:53 2022 daemon.crit dnsmasq[1]: FAILED to start up
Sat Oct 22 21:30:58 2022 daemon.crit dnsmasq[1]: cannot access directory /tmp/dnsmasq.ssr: No such file or directory
Sat Oct 22 21:30:58 2022 daemon.crit dnsmasq[1]: FAILED to start up
Sat Oct 22 21:31:02 2022 daemon.err odhcpd[5690]: Failed to send to ff02::1%lan@br-lan (Bad file descriptor)
Sat Oct 22 21:31:03 2022 daemon.crit dnsmasq[1]: cannot access directory /tmp/dnsmasq.ssr: No such file or directory
Sat Oct 22 21:31:03 2022 daemon.crit dnsmasq[1]: FAILED to start up
Sat Oct 22 21:31:08 2022 daemon.crit dnsmasq[1]: cannot access directory /tmp/dnsmasq.ssr: No such file or directory
Sat Oct 22 21:31:08 2022 daemon.crit dnsmasq[1]: FAILED to start up
Sat Oct 22 21:31:13 2022 daemon.crit dnsmasq[1]: cannot access directory /tmp/dnsmasq.ssr: No such file or directory
Sat Oct 22 21:31:13 2022 daemon.crit dnsmasq[1]: FAILED to start up
Sat Oct 22 21:31:18 2022 daemon.crit dnsmasq[1]: cannot access directory /tmp/dnsmasq.ssr: No such file or directory
Sat Oct 22 21:31:18 2022 daemon.crit dnsmasq[1]: FAILED to start up
Sat Oct 22 21:31:18 2022 daemon.info procd: Instance dnsmasq::cfg01411c s in a crash loop 6 crashes, 0 seconds since last crash

DHDAXCW, 2022-10-23:
这hw搞得头大了...

quango-web, 2022-10-23:
老师傅碰到疑难杂症了。。。。

DHDAXCW, 2022-10-25:
https://github.com/DHDAXCW/NanoPi-R4S-R4SE/releases/tag/2022.10.25-Lean3

quango-web, 2022-10-25:
已使用10.25全固件版本的HW，之前版本出现的故障已得到解决，暂无其他反馈！

closed 2022-10-25 by zonyitoo