DHDAXCW/NanoPi-R4S-R4SE #288: Action现在编译完出一大堆警告怎么改？

qqsir-dev, 2022-10-28:
<html>
<body>
<div class="mt-1 mb-3 px-3" style="box-sizing: border-box; margin-top: 4px !important; margin-bottom: 16px !important; padding-right: 16px !important; padding-left: 16px !important; color: rgb(36, 41, 47); font-family: -apple-system, BlinkMacSystemFont, &quot;Segoe UI&quot;, Helvetica, Arial, sans-serif, &quot;Apple Color Emoji&quot;, &quot;Segoe UI Emoji&quot;; font-size: 14px; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; white-space: normal; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;"><div class="mx-0 mx-md-1" style="box-sizing: border-box; margin-right: 4px !important; margin-left: 4px !important;"><h4 class="text-bold" style="box-sizing: border-box; margin-top: 0px; margin-bottom: 0px; font-size: 16px; font-weight: 600 !important;">Annotations</h4><div class="color-fg-muted text-small" style="box-sizing: border-box; color: var(--color-fg-muted)  !important; font-size: 12px !important;">15 warnings</div></div></div>

BuildNode.js 12 actions are deprecated. For more information see: https://github.blog/changelog/2022-09-22-github-actions-all-actions-will-begin-running-on-node16-instead-of-node12/. Please update the following actions to use Node.js 16: softprops/action-gh-release, GitRML/delete-workflow-runs, dev-drprasad/delete-older-releases, geekyeggo/delete-artifactShow more
--
BuildThe `set-output` command is deprecated and will be disabled soon. Please upgrade to using Environment Files. For more information see: https://github.blog/changelog/2022-10-11-github-actions-deprecating-save-state-and-set-output-commands/Show more
BuildThe `set-output` command is deprecated and will be disabled soon. Please upgrade to using Environment Files. For more information see: https://github.blog/changelog/2022-10-11-github-actions-deprecating-save-state-and-set-output-commands/Show more
BuildThe `set-output` command is deprecated and will be disabled soon. Please upgrade to using Environment Files. For more information see: https://github.blog/changelog/2022-10-11-github-actions-deprecating-save-state-and-set-output-commands/Show more
BuildThe `set-output` command is deprecated and will be disabled soon. Please upgrade to using Environment Files. For more information see: https://github.blog/changelog/2022-10-11-github-actions-deprecating-save-state-and-set-output-commands/Show more
BuildThe `set-output` command is deprecated and will be disabled soon. Please upgrade to using Environment Files. For more information see: https://github.blog/changelog/2022-10-11-github-actions-deprecating-save-state-and-set-output-commands/Show more
BuildThe `set-output` command is deprecated and will be disabled soon. Please upgrade to using Environment Files. For more information see: https://github.blog/changelog/2022-10-11-github-actions-deprecating-save-state-and-set-output-commands/Show more
BuildThe `set-output` command is deprecated and will be disabled soon. Please upgrade to using Environment Files. For more information see: https://github.blog/changelog/2022-10-11-github-actions-deprecating-save-state-and-set-output-commands/Show more
BuildThe `set-output` command is deprecated and will be disabled soon. Please upgrade to using Environment Files. For more information see: https://github.blog/changelog/2022-10-11-github-actions-deprecating-save-state-and-set-output-commands/Show more
BuildThe `set-output` command is deprecated and will be disabled soon. Please upgrade to using Environment Files. For more information see: https://github.blog/changelog/2022-10-11-github-actions-deprecating-save-state-and-set-output-commands/Show more
BuildThe `set-output` command is deprecated and will be disabled soon. Please upgrade to using Environment Files. For more information see: https://github.blog/changelog/2022-10-11-github-actions-deprecating-save-state-and-set-output-commands/Show more
BuildUnable to delete artifact "bin-archive"; the artifact was not found.
BuildThe `set-output` command is deprecated and will be disabled soon. Please upgrade to using Environment Files. For more information see: https://github.blog/changelog/2022-10-11-github-actions-deprecating-save-state-and-set-output-commands/Show more
BuildThe `set-output` command is deprecated and will be disabled soon. Please upgrade to using Environment Files. For more information see: https://github.blog/changelog/2022-10-11-github-actions-deprecating-save-state-and-set-output-commands/Show more
BuildThe `set-output` command is deprecated and will be disabled soon. Please upgrade to using Environment Files. For more information see: https://github.blog/changelog/2022-10-11-github-actions-deprecating-save-state-and-set-output-commands/

</body>
</html>Annotations
15 warnings
[Build](https://github.com/DHDAXCW/NanoPi-R4S-R4SE/actions/runs/3341971986/jobs/5533711573)
Node.js 12 actions are deprecated. For more information see: https://github.blog/changelog/2022-09-22-github-actions-all-actions-will-begin-running-on-node16-instead-of-node12/. Please update the following actions to use Node.js 16: softprops/action-gh-release, GitRML/delete-workflow-runs, dev-drprasad/delete-older-releases, geekyeggo/delete-artifact
[Build](https://github.com/DHDAXCW/NanoPi-R4S-R4SE/actions/runs/3341971986/jobs/5533711573#step:11:22)
The `set-output` command is deprecated and will be disabled soon. Please upgrade to using Environment Files. For more information see: https://github.blog/changelog/2022-10-11-github-actions-deprecating-save-state-and-set-output-commands/
[Build](https://github.com/DHDAXCW/NanoPi-R4S-R4SE/actions/runs/3341971986/jobs/5533711573#step:11:23)
The `set-output` command is deprecated and will be disabled soon. Please upgrade to using Environment Files. For more information see: https://github.blog/changelog/2022-10-11-github-actions-deprecating-save-state-and-set-output-commands/
[Build](https://github.com/DHDAXCW/NanoPi-R4S-R4SE/actions/runs/3341971986/jobs/5533711573#step:11:24)
The `set-output` command is deprecated and will be disabled soon. Please upgrade to using Environment Files. For more information see: https://github.blog/changelog/2022-10-11-github-actions-deprecating-save-state-and-set-output-commands/
[Build](https://github.com/DHDAXCW/NanoPi-R4S-R4SE/actions/runs/3341971986/jobs/5533711573#step:9:878)
The `set-output` command is deprecated and will be disabled soon. Please upgrade to using Environment Files. For more information see: https://github.blog/changelog/2022-10-11-github-actions-deprecating-save-state-and-set-output-commands/
[Build](https://github.com/DHDAXCW/NanoPi-R4S-R4SE/actions/runs/3341971986/jobs/5533711573#step:9:879)
The `set-output` command is deprecated and will be disabled soon. Please upgrade to using Environment Files. For more information see: https://github.blog/changelog/2022-10-11-github-actions-deprecating-save-state-and-set-output-commands/
[Build](https://github.com/DHDAXCW/NanoPi-R4S-R4SE/actions/runs/3341971986/jobs/5533711573#step:9:880)
The `set-output` command is deprecated and will be disabled soon. Please upgrade to using Environment Files. For more information see: https://github.blog/changelog/2022-10-11-github-actions-deprecating-save-state-and-set-output-commands/
[Build](https://github.com/DHDAXCW/NanoPi-R4S-R4SE/actions/runs/3341971986/jobs/5533711573#step:9:881)
The `set-output` command is deprecated and will be disabled soon. Please upgrade to using Environment Files. For more information see: https://github.blog/changelog/2022-10-11-github-actions-deprecating-save-state-and-set-output-commands/
[Build](https://github.com/DHDAXCW/NanoPi-R4S-R4SE/actions/runs/3341971986/jobs/5533711573#step:9:882)
The `set-output` command is deprecated and will be disabled soon. Please upgrade to using Environment Files. For more information see: https://github.blog/changelog/2022-10-11-github-actions-deprecating-save-state-and-set-output-commands/
[Build](https://github.com/DHDAXCW/NanoPi-R4S-R4SE/actions/runs/3341971986/jobs/5533711573#step:10:50)
The `set-output` command is deprecated and will be disabled soon. Please upgrade to using Environment Files. For more information see: https://github.blog/changelog/2022-10-11-github-actions-deprecating-save-state-and-set-output-commands/ [... cut ...]
The `set-output` command is deprecated and will be disabled soon. Please upgrade to using Environment Files. For more information see: https://github.blog/changelog/2022-10-11-github-actions-deprecating-save-state-and-set-output-commands/
[Build](https://github.com/DHDAXCW/NanoPi-R4S-R4SE/actions/runs/3341971986/jobs/5533711573#step:12:37)
The `set-output` command is deprecated and will be disabled soon. Please upgrade to using Environment Files. For more information see: https://github.blog/changelog/2022-10-11-github-actions-deprecating-save-state-and-set-output-commands/

DHDAXCW, 2022-10-28:
这种警告你可以无视就行，反正保持到2023年，到2023年再改吧

DHDAXCW, 2022-10-28:
https://github.com/DHDAXCW/NanoPi-R4S-R4SE/commit/9c06d984f44aad749c50e57305231fe5c5cb7034  参考我这个改吧。。。

qqsir-dev, 2022-10-28:
> https://github.com/DHDAXCW/NanoPi-R4S-R4SE/commit/9c06d984f44aad749c50e57305231fe5c5cb7034  参考我这个改吧。。。

已经照着你的改了😁

DHDAXCW, 2022-10-29:
https://github.com/DHDAXCW/NanoPi-R4S-R4SE/commit/3eaf0aba87180e058ca82cb4bfe29dcff62655e9  再补上~

qqsir-dev, 2022-10-29:
嗯，改了也还有警告，实在不知道怎么才能屏蔽有一个node js 12的老也去不掉，那几个action/也不知道怎么更新。。

DHDAXCW, 2022-10-29:
我也出现这样，最后那个搞不懂在哪里可以改。。。。。懵逼了

closed 2022-11-01 by DHDAXCW