DHDAXCW/NanoPi-R4S-R4SE #293: 云编译出错

qqsir-dev, 2022-11-19:
make[2] -C package/kernel/mac80211 compile
    ERROR: package/kernel/mac80211 failed to build (build variant: regular).
make -r package/compile: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
make: *** [/home/runner/work/AutoLede/AutoLede/openwrt/include/toplevel.mk:230: package/compile] Error 1

去掉mac80211 kmod还是一样出这个错误

DHDAXCW, 2022-11-19:
日志不完善，看不出来。。

qqsir-dev, 2022-11-20:
from ./include/linux/etherdevice.h:20,
                 from /home/runner/work/AutoLede/AutoLede/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-5.15.33-1/backport-include/linux/etherdevice.h:3,
                 from /home/runner/work/AutoLede/AutoLede/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-5.15.33-1/drivers/net/wireless/marvell/libertas/rx.c:8:
/home/runner/work/AutoLede/AutoLede/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-5.15.33-1/include/linux/overflow.h:32: warning: "is_signed_type" redefined
   32 | #define is_signed_type(type)       (((type)(-1)) < (type)1)
      | 
In file included from /home/runner/work/AutoLede/AutoLede/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-5.15.33-1/backport-include/linux/compiler.h:3,
                 from ./include/linux/export.h:33,
                 from ./include/linux/linkage.h:7,
                 from ./include/linux/kernel.h:17,
                 from /home/runner/work/AutoLede/AutoLede/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-5.15.33-1/backport-include/linux/kernel.h:3,
                 from ./include/linux/skbuff.h:13,
                 from /home/runner/work/AutoLede/AutoLede/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-5.15.33-1/backport-include/linux/skbuff.h:3,
                 from ./include/linux/if_ether.h:19,
                 from /home/runner/work/AutoLede/AutoLede/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-5.15.33-1/backport-include/linux/if_ether.h:3,
                 from ./include/linux/etherdevice.h:20,
                 from /home/runner/work/AutoLede/AutoLede/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-5.15.33-1/backport-include/linux/etherdevice.h:3,
                 from /home/runner/work/AutoLede/AutoLede/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-5.15.33-1/drivers/net/wireless/marvell/libertas/rx.c:8:
./include/linux/compiler.h:248: note: this is the location of the previous definition
  248 | #define is_signed_type(type) (((type)(-1)) < (__force type)1)
      | 
/home/runner/work/AutoLede/AutoLede/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-5.15.33-1/drivers/net/wireless/marvell/libertas/rx.c: In function 'lbs_process_rxed_packet':
/home/runner/work/AutoLede/AutoLede/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-5.15.33-1/drivers/net/wireless/marvell/libertas/rx.c:150:9: error: implicit declaration of function 'netif_rx_any_context' [-Werror=implicit-function-declaration]
  150 |         netif_rx_any_context(skb);
      |         ^~~~~~~~~~~~~~~~~~~~
cc1: some warnings being treated as errors
make[10]: *** [scripts/Makefile.build:249: /home/runner/work/AutoLede/AutoLede/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-5.15.33-1/drivers/net/wireless/marvell/libertas/rx.o] Error 1
make[9]: *** [scripts/Makefile.build:465: /home/runner/work/AutoLede/AutoLede/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-5.15.33-1/drivers/net/wireless/marvell/libertas] Error 2
make[8]: *** [scripts/Makefile.build:465: /home/runner/work/AutoLede/AutoLede/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-5.15.33-1/drivers/net/wireless/marvell] Error 2
make[7]: *** [scripts/Makefile.build:465: /home/runner/work/AutoLede/AutoLede/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-5.15.33-1/drivers/net/wireless] Error 2
make[6]: *** [Makefile:1852: /home/runner/work/AutoLede/AutoLede/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-5.15.33-1] Error 2
make[5]: *** [Makefile.build:13: modules] Error 2
make[4]: *** [Makefile.real:93: modules] Error 2
make[3]: *** [Makefile:121: modules] Error 2
make[3]: Leaving directory '/home/runner/work/AutoLede/AutoLede/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-5.15.33-1'
make[2]: *** [Makefile:559: /home/runner/work/AutoLede/AutoLede/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-5.15.33-1/.built] Error 2
make[2]: Leaving directory '/home/runner/work/AutoLede/AutoLede/openwrt/package/kernel/mac80211'
time: package/kernel/mac80211/regular/compile#1.19#0.31#1.44
    ERROR: package/kernel/mac80211 failed to build (build variant: regular).
make[1]: *** [package/Makefile:116: package/kernel/mac80211/compile] Error 1
make[1]: Leaving directory '/home/runner/work/AutoLede/AutoLede/openwrt'
make: *** [/home/runner/work/AutoLede/AutoLede/openwrt/include/toplevel.mk:230: package/compile] Error 2
Error: Process completed with exit code 2.

closed 2023-02-16 by DHDAXCW