DHDAXCW/NanoPi-R4S-R4SE #312: 做拨号主路由经常早上起来晚上断网，发现只亮红灯，表面pwr红灯亮，其他灯都不亮，网线插口那边的wan lan接口都亮

hqwuzhaoyi, 2023-02-22:
做拨号主路由经常早上起来晚上断网，发现只亮红灯，表面pwr红灯亮，其他灯都不亮，网线插口那边的wan lan接口都亮

下面是日志
```
Feb 22 00:00:00 FusionWrt crond[5981]: USER root pid 4944 cmd /usr/share/AdGuardHome/tailto.sh 2000 $(uci get AdGuardHome.AdGuardHome.workdir)/data/querylog.json
Feb 22 00:00:00 FusionWrt crond[5981]: USER root pid 4945 cmd /usr/bin/serverchan/serverchan send &
Feb 22 00:00:01 FusionWrt crond[5157]: crond (busybox 1.36.0) started, log level 8
Feb 22 01:00:00 FusionWrt crond[5157]: USER root pid 407 cmd /usr/share/AdGuardHome/tailto.sh 2000 $(uci get AdGuardHome.AdGuardHome.workdir)/data/querylog.json
Feb 22 01:01:00 FusionWrt crond[5157]: USER root pid 7249 cmd sh /usr/share/jd-dailybonus/newapp.sh -w
Feb 22 02:00:00 FusionWrt crond[5157]: USER root pid 18214 cmd /usr/share/AdGuardHome/tailto.sh 2000 $(uci get AdGuardHome.AdGuardHome.workdir)/data/querylog.json
Feb 22 02:31:12 FusionWrt netdata[27536]: Initializing file /var/cache/netdata/net_drops.eth1/main.db.
Feb 22 02:31:12 FusionWrt netdata[27536]: Initializing file /var/cache/netdata/net_drops.eth1/inbound.db.
Feb 22 02:31:12 FusionWrt netdata[27536]: Initializing file /var/cache/netdata/net_drops.eth1/outbound.db.
Feb 22 03:00:00 FusionWrt crond[5157]: USER root pid 10364 cmd /usr/share/AdGuardHome/tailto.sh 2000 $(uci get AdGuardHome.AdGuardHome.workdir)/data/querylog.json
Feb 22 03:30:00 FusionWrt crond[5157]: USER root pid 29511 cmd /usr/share/AdGuardHome/update_core.sh 2>&1
Feb 22 03:30:00 FusionWrt crond[5157]: USER root pid 29512 cmd /usr/share/AdGuardHome/tailto.sh 2000 $(uci get AdGuardHome.AdGuardHome.logfile)
Feb 22 03:30:00 FusionWrt crond[5157]: USER root pid 29513 cmd /usr/share/AdGuardHome/gfw2adg.sh
Feb 22 03:30:13 FusionWrt procd: instance nobackup not found
Feb 22 03:31:00 FusionWrt crond[5157]: USER root pid 2925 cmd /usr/share/AdGuardHome/gfwipset2adg.sh

```

closed 2023-02-25 by hqwuzhaoyi

hardass, 2023-04-23:
用最新的固件也发现同样的问题

DHDAXCW, 2023-04-23:
再提供一下内核日志

hardass, 2023-04-24:
> 再提供一下内核日志

只亮红灯的时候，已经无法登陆。只能拔电源重启。会丢失内存日志。我看了一下/var/log里面并没有syslog的输出。
现在我用内网另外一台机器ssh到r4s上，然后tail -f /dev/kmsg。希望可以等到问题重现的时候能隔空捕捉到些信息。如果有结果了我会回来回复。
当然如果有更好的监控方法也请告知。谢谢。

hardass, 2023-04-26:
> 再提供一下内核日志

今晚又重现了一次问题，r4s，另一台机器ssh去监控kmsg没有得到任何更新就直接丢失了r4s的连接了。
请教一下如何能有效的捕捉内核错误。谢谢。