DHDAXCW/NanoPi-R4S-R4SE #320: 大神何時會更新SmartDNS到R4S的固件里

wslftx, 2023-05-21:
我下载了最新的R4S Slim版本 2023-04-28，可惜里面还是没有我需要的SmartDNS （配合我个人的VPS使用，效果很好），不知大神何时会更新SmartDNS到新固件里？

我一直用的是大神编译的2021-11-10这个版本，有些插件过时了需要更新，但是新固件没有我需要的SmartDNS，很是纠结。。。
DHDAXCW @ FusionWrt 20211110 R21.11.11 (2021-11-10) / LuCI Master (git-21.280.14233-9b796b1)

GreenTeodoro839, 2023-05-21:
自己安装就行了

wslftx, 2023-05-22:
我有按照SmartDNS网站的说明（如下）试着安装过，在系统 - 软件包-已安装软件包中可以找到这个刚安装的SmartDNS版本，但是在“服务”菜单里却找不到SmartDNS这一项，没法进行相应的设置。不知道哪里出问题了。还请大神指点迷津！

OpenWrt安装
将软件包（使用 WinSCP 等）上传到路由器的 /root 目录，执行如下命令安装
opkg install smartdns.1.yyyy.MM.dd-REL.xxxx.ipk
opkg install luci-app-smartdns.1.yyyy.MM.dd-REL.all.ipk
注意： 19.07 之前的版本，请务必安装 luci-app-smartdns.1.yyyy.MM.dd-REL.all-luci-compat-all.ipk。
修改配置
登录 OpenWrt 管理页面，打开 Services -> SmartDNS 进行配置。